bolliy/ioBroker.sun2000 #16: Why changing something without user interaction?

ansgarschulte, 2024-01-02:
https://github.com/bolliy/ioBroker.sun2000/blob/5bf048e03a261e901d6807dc3f2995880b376367/main.js#L156C51-L156C51

Some question here:

1.  I have a LUNA2000 and my storageModel is 0?!
2. Why do you change the config to this?

bolliy, 2024-01-02:
It is the preparation for being able to obtain cheap stock market electricity via tibber on a time-controlled basis. But I will comment it out in the next version. You can easily switch the working mode from "TOU" to "maximum self-consuption" via the app.
Then I don't understand the statement/question about storageModel 0? The storage/battery LUNA2000 has the value of storageModel 2.

ansgarschulte, 2024-01-03:
I debugged the adapter locally with my storage... and I get a storageModel == 0 instead of 2. And I have a LUNA2000 connected.
[image]

bolliy, 2024-01-04:
The modbus definition description (https://www.photovoltaikforum.com/core/attachment/315028-solar-inverter-modbus-interface-definitions-pdf/ Page 64) is written the value 2 in register 47089 for LUNA2000. Maybe something is configured incorrectly?

bolliy, 2024-01-12:
Can I close the issue? This function is disabled in the current version.

ansgarschulte, 2024-01-15:
yes

closed 2024-01-15 by ansgarschulte