bolliy/ioBroker.sun2000 #22: huawei optimizer

gschlad, 2024-01-12:
include data from optimizer.

bolliy, 2024-01-12:
Can you define exactly which optimizer data you need? I don't have any optimizers. That's why I need your support. https://forum.iobroker.net/assets/uploads/files/1699119419919-solar-inverter-modbus-interface-definitions-v5.pdf

gschlad, 2024-01-12:
they simple provide interessting data and allow you to monitor the performence for every single module:

    Input Current
    Input Voltage
    Output Current
    Output Voltage
    Output Power
    Temperature
    Accumulated Energy Yield
    Running Status (online/offline/disconnected)

[image]

there are some solutions what already provide this data => https://pypi.org/project/huawei-solar/ 

the problem is that it´s not a simpler modbus register, check the page 121 (6.3.7.1 Uploading Files)  https://forum.iobroker.net/assets/uploads/files/1699119419919-solar-inverter-modbus-interface-definitions-v5.pdf

bolliy, 2024-01-12:
OK, I'll try to implement it. How many inverters and optimizers do you use? 
If there are many optimzers, performance problems can occur when reading out the register data.

baerengraben, 2024-01-12:
I'm interested in this feature too and I could help with tests. 
I do have 2 Inverters and 58 optimizers.

gschlad, 2024-01-12:
now i have 29 optimizer with one inverter, more to come.

so far i had no problem in the 7 months, i already use them in home assistent. I use the dongle for ioBroker and the integrated accesspoint with home assistent.

bolliy, 2024-10-29:
In my opinion, effort is not proportional to success. Someone wanted to help me with the implementation, but they dropped out.
That's why I decided not to implement this in the adapter.

Stephan

baerengraben, 2024-11-02:
I have briefly studied the Huawei documentation and it seems that it is not really complete. The best way might be to rewrite the Python library (above) through reverse engineering. Although I would help with that, since I am not an expert in either Python or JavaScript, this would mean too much effort for me too.
So, probably the fastest but really ugly way would be simply to create an own JS script that uses the Python library and adds the additional objects to your adapter. This could work with the proxy-function of your adapter.

Ikswoknib, 2025-03-19:
I would also be very interested in data from the optimizers and could help with testing. Maybe i could help a bit with programming but i am not a pro in JS...

Towi68, 2025-03-26:
I'm also very interested in the data from this optimizers.
Today I'm read the normal registers with modbus-tcp.
But read of the data from the optimizers looks more complex.
Any help are welcome.