bolliy/ioBroker.sun2000 #35: Verbindung Problem bzw. nicht stabile

gmihalca, 2024-01-23:
[image]

Hi,  als erstens vielen dank für den tollen Iobroker Adapter

schaffe den aber nicht mehr als 12h in Betrieb zu halten es verliert immer wieder die verbindung und bekomme die Meldung
 Error while reading form 192.168.0.10 (IP Dongle):...... whit :Port Not Open

Hab 2 x Huawei 1x10kw 1x6kw 2xLuna je 10 Kw alles mit dem letztem Update 
Softwareversion
V200R022C10SPC108
 

Wechselrichter | SUN2000-6KTL-M1 | V100R001C00SPC159 

Wechselrichter | SUN2000-10KTL-M1 | V100R001C00SPC159 

Mit Note-Red bleibt die Verbindung aufrecht
und auch mit dem Adapter von daolis (da wird aber nichts weiter gemacht und es kann auch nur ein WR abgefragt werden)

bolliy, 2024-01-23:
Erstmal vielen Dank, dass du den Adapter einsetzt und testet!
Welche Version setzt du gerade ein?
Ich werde in den nächsten Tagen ein Update veröffentlichen. Vielleicht können wir auf Basis der neuen Version nach einer Lösung suchen.
LG Stephan

gmihalca, 2024-01-23:
hi, ich hab die letzte Version v0.1.3 in Verwendung 

lg
gmihalca

bolliy, 2024-01-25:
Könnte dein WR im Standby Mode sein? Dann kann entweder keine Verbindung zum Dongle aufgebaut werden oder einige Wert von der Batterie stehen nicht mehr zur Verfügung. Die Zeit Zeit 19.30h würde zu dem Verhalten passen. 
Siehe #34
Hast du die neue Version 0.2.0 getestet?

gmihalca, 2024-01-25:
hi, die neue Version 0.2.0 ist seit gestern im Test und es lauft super bis jetzt.

viele Dank.

gmihalca, 2024-01-25:
sollte für mich sich erledigt haben, vielen Dank nochmals.

commit 9758b67d (2024-01-23): try to fix #35
closed 2024-01-25 by gmihalca