bolliy/ioBroker.sun2000 #41: SmartLogger3000

Kaulii, 2024-01-28:
Implementierung vom SmartLogger 3000.
Der Zähler hat die Modbus Adresse 11.  Dieser hängt nicht direkt am Wechselrichter, sondern am SL3000.
WR1 = Modbus Adresse 1, WR2 = Modbus Adresse 2, Zähler DTSU666 = Modbus Adresse 11
WR1 hat den Batteriespeicher unter sich. hier bekomme ich mit dem Adapter auch schon Werte.
Falls ich irgendwas zur verfügung stellen kann/darf/soll. Einfach melden.

Lg Michael
[image]

bolliy, 2024-02-03:
Hallo Michael,

da ich gerade mit der Umstrukturierung des Adapter befasst bin, werde ich mich vorest nicht mit der Umsetzung befassen.

LG Stephan

bolliy, 2024-02-10:
Hallo Michael, ich habe den Adapter so umgebaut, dass ich ggf. den smartLogger300 implementieren kann.
Siehe: https://forum.iobroker.net/topic/71768/test-adapter-sun2000-v0-1-x-huawei-wechselrichter

Würest du die neue **V0.03** installieren und den SmartLogger3000 als ersten inverter ansprechen.
Also **mobus Inverter IDs : 11,1,2** anpassen. 
Im Log und im State **sun2000.0.inverter.0.info.modelID** nach der Model ID suchen. Die benötige ich um die Hardware zu identifizieren. 
Hast du die Modbus Definition Definitions von dem SmartLogger3000 für mich?

Und dann bin auf deine Reaktion gespannt.
LG Stephan

Kaulii, 2024-02-12:
Hallo Stephan,

habe es getestet, der SmartLogger3000 hat nicht die ID11, sondern die ID 0, dieser wird über den Adressbereich am 40000 angesprochen, siehe hier die Modbus Definition https://support.huawei.com/enterprise/en/doc/EDOC1100050690 somit erhält er im log unter info auch keinen Wert.
[image]

zum Bild: hier hab ich es mit adresse 11 Probiert. 

Wenn ich es mit 0,1,2 teste habe ich die gleiche Meldung.
Seltsam finde ich das er den Wechselrichter mit der Adress 1 dann aber auch nicht auslesen kann. Wechselrichter mit der Adresse 2 hat alle werte im Log. Als Fehlermeldung zeigt er folgendes an:
[image]

Wenn ich nur die beiden Wechselrichter eintrage  **modbus Inverter IDs: 1,2**  erhält er folgende Meldung:
[image]

Die Meldung das nur 1 Client erlaubt ist, hatte ich mit der Version V.0.2.0 nicht.

LG Michael

.

bolliy, 2024-02-14:
Ich werde zuerst den SmartCharger in den Adapter aufnehmen, da dieser deutlich einfachere States hat.
Stephan

lcars79, 2024-02-28:
Hallo Kaulii,
eine Frage zum SmartLogger3000 den du sicherlich hast. Wie schnell kannst du da momentan die Daten abfragen. Hintergrund ist der, dass der Dongle da etwas träge ist und z.T. mit Timeouts antwortet. Gibt es da einen großen Unterschied zum SmartLogger oder ist das vom Abruf her ähnlich?
Momentan habe ich es geschafft bei mir ein Abrufintervall von 10s einzustellen und er braucht dann ca. 15s für einen Durchlauf laut Logs. Ich denke das ist schon hart an der Grenze.

Danke für deine Antwort.

Gruß
Frank

Kaulii, 2024-03-01:
Hallo Frank,

ich kann den Abrufintervall auf 5s einstellen. (kleiner aktuell nicht möglich) bekomme hier auch keine Timeouts.
Der SL3000 wird bestimmt etwas schneller sein.
Denke es ist aber auch abhängig wieviel Geräte ausgelesen werden.
Zu meiner Konfiguration: 1x 10klM1 +20 Optimierer  +1x Luna mit 15kwh und 1x 20ktlM2 + 54 Optimierer (sprich 2x WR)
Jedoch funktioniert der Adapter bei mir noch nicht ganz. Die Werte des Meters kann ich aktuell nicht auswerten.
Der Zähler DTSU666 muss in der Konfiguration mit dem SL3000 direkt am SL3000 angeschlossen werden.
Somit kann der Zähler nur über die eingestellte Adresse (11) asgelesen werden.

@bolliy habe mit der konifguration noch einmal ein bisschen gespielt.
mit der Konfigurationsreihenfolge:
[image]
habe ich keine eintragungen im Log mehr wie im obrigen Post geschriebenen.

[image]

Gruß 
Michael

lcars79, 2024-03-01:
Danke für die Info.

bolliy, 2024-03-01:
Hi Michael,
deinen Logger wollte ich auch noch implementieren. Jetzt werde zuerst die v0.4.0 auf npm veröffentlichen.
Ich hoffe, dass ich bald dazu komme...
Stephan

bolliy, 2024-03-04:
Hi Michael,
habe nun eine Test-Version hochgeladen, die dem SmartLogger bedienen soll!
Da ich keinen SmartLogger habe ist das natürlich ein Blindflug. Aber ich hoffe, dass es dennoch funktioniert.

Kannst also über **gitHub** einfache installlieren. So ähnlich müsste deine Settings aussehen. (Ip-Adresse muss aber angepasst werden)

VG Stephan

[image]

Kaulii, 2024-03-04:
Hi,

super Arbeit!!
Bin gerade am Testen.
Es treten noch ein paar Fehler auf, aber die Werte vom Zähler sind größtenteils da.
[image]
im ordner Slogger wird nichts angezeigt. der Smartlogger hat die ID 0, falls du hier auch irgendwelche Werte auslesen würdest?

Was wird über die ID 100 abgefragt?
[image]

Die 40007 muss vom SL3000 über die ID 0 abgefragt werden.
Error while reading from 192.168.3.108 [Reg: 40007, Len: 1, modbusID: 100] with: Port Not Open

Werde es morgen Tagsüber weiter beobachten und mich hier Melden.

Danke schonmal für deine Mühe :-)

Kaulii, 2024-03-04:
[image]

Kaulii, 2024-03-04:
Der Wert "sun2000.0.meter.accumulatedReactivePowe" im Adressbereich 32361  steht auf 0kwh, wenn ich den Wert Auslese bekomme ich folgenden Wert: 
[image]
Liegt es vlt an dem 64Bit Integer Wert?

Kaulii, 2024-03-05:
Hi,

Werte vom Messgerät sind jetzt alle da! (also auch die 64Bit Werte)
[image]
unter Slogger passt schonmal die Anzahl der Welchselrichter :-)
Der Wert vom 42256 habe ich in meiner Modbusregister Tabelle für den SL3000 nicht. Hast du eine andere Liste?
Im Log hab ich noch folgende Meldungen:

<html><body>

sun2000.0 | 2024-03-05 11:09:17.246 | info | Connected Modbus TCP to 192.168.3.108:502
-- | -- | -- | --
sun2000.0 | 2024-03-05 11:09:12.242 | info | Open Connection...
sun2000.0 | 2024-03-05 11:09:12.242 | warn | Error while reading from 192.168.3.108 [Reg: 40007, Len: 1, modbusID: 0] with: Port Not Open
sun2000.0 | 2024-03-05 11:09:12.241 | warn | Consider that only 1 client is allowed to connect to modbus at the same time!
sun2000.0 | 2024-03-05 11:09:12.241 | warn | Has another device interrupted the modbus connection?
sun2000.0 | 2024-03-05 11:09:12.016 | warn | Error  while reading from 192.168.3.108 [Reg: 32066, Len: 51, modbusID: 2]  with: Modbus exception 3: Illegal data value (value cannot be written to  this register)
sun2000.0 | 2024-03-05 11:09:11.616 | info | Connected Modbus TCP to 192.168.3.108:502
sun2000.0 | 2024-03-05 11:09:06.611 | info | Open Connection...
sun2000.0 | 2024-03-05 11:09:06.610 | warn | Error while reading from 192.168.3.108 [Reg: 37758, Len: 30, modbusID: 1] with: Port Not Open
sun2000.0 | 2024-03-05 11:09:06.610 | warn | Consider that only 1 client is allowed to connect to modbus at the same time!
sun2000.0 | 2024-03-05 11:09:06.609 | warn | Has another device interrupted the modbus connection?
sun2000.0 | 2024-03-05 11:09:06.219 | warn | Error  while reading from 192.168.3.108 [Reg: 32066, Len: 51, modbusID: 1]  with: Modbus exception 3: Illegal data value (value cannot be written to  this register)
sun2000.0 | 2024-03-05 11:09:05.414 | info | Adjustment: It has reached the step 2 with delay 488 ms
sun2000.0 | 2024-03-05 11:09:05.005 | info | Connected Modbus TCP to 192.168.3.108:502
sun2000.0 | 2024-03-05 11:09:00.001 | info | Open Connection...
sun2000.0 | 2024-03-05 11:08:49.244 | warn | Error  while reading from 192.168.3.108 [Reg: 32066, Len: 51, modbusID: 2]  with: Modbus exception 3: Illegal data value (value cannot be written to  this register)
sun2000.0 | 2024-03-05 11:08:47.685 | info | Connected Modbus TCP to 192.168.3.108:502
sun2000.0 | 2024-03-05 11:08:42.678 | info | Open Connection...
sun2000.0 | 2024-03-05 11:08:42.677 | warn | Error while reading from 192.168.3.108 [Reg: 37758, Len: 30, modbusID: 1] with: Port Not Open
sun2000.0 | 2024-03-05 11:08:42.676 | warn | Consider that only 1 client is allowed to connect to modbus at the same time!
sun2000.0 | 2024-03-05 11:08:42.674 | warn | Has another device interrupted the modbus connection?
sun2000.0 | 2024-03-05 11:08:42.576 | warn | Error  while reading from 192.168.3.108 [Reg: 32066, Len: 51, modbusID: 1]  with: Modbus exception 3: Illegal data value (value cannot be written to  this register)
sun2000.0 | 2024-03-05 11:08:41.877 | info | Adjustment: It has reached the step 1 with delay 600 ms
sun2000.0 | 2024-03-05 11:08:35.437 | info | Connected Modbus TCP to 192.168.3.108:502
sun2000.0 | 2024-03-05 11:08:30.432 | info | Open Connection...
sun2000.0 | 2024-03-05 11:08:30.413 | warn | Error while reading from 192.168.3.108 [Reg: 37200, Len: 3, modbusID: 1] with: Timed out
sun2000.0 | 2024-03-05 11:08:20.407 | warn | Error  while reading from 192.168.3.108 [Reg: 32066, Len: 51, modbusID: 1]  with: Modbus exception 3: Illegal data value (value cannot be written to  this register)
sun2000.0 | 2024-03-05 11:08:20.117 | info | Adjustment: It has reached the step 2 with delay 0 ms
sun2000.0 | 2024-03-05 11:08:08.329 | info | Adjustment: It has reached the step 1 with delay 0 ms
sun2000.0 | 2024-03-05 11:08:07.931 | info | Connected Modbus TCP to 192.168.3.108:502
sun2000.0 | 2024-03-05 11:08:02.923 | info | Open Connection...
sun2000.0 | 2024-03-05 11:08:02.908 | info | Adjustment: It starts for the Modbus connection...
sun2000.0 | 2024-03-05 11:08:02.655 | info | starting.  Version 0.4.1 (non-npm: bolliy/ioBroker.sun2000) in  /opt/iobroker/node_modules/iobroker.sun2000, node: v18.19.0,  js-controller: 5.0.17
host.iobroker | 2024-03-05 11:08:00.253 | info | instance system.adapter.sun2000.0 started with pid 49402
host.iobroker | 2024-03-05 11:08:00.194 | info | "system.adapter.sun2000.0" enabled
host.iobroker | 2024-03-05 11:07:12.687 | info | instance system.adapter.sun2000.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
sun2000.0 | 2024-03-05 11:07:12.631 | info | terminating
sun2000.0 | 2024-03-05 11:07:12.567 | info | Open Connection...

</body>
</html>

Kaulii, 2024-03-05:
vom 20KTL werden die Optimierer nicht mehr angezeigt.
Ging in der Version 0.3.0 meine ich noch.
schöne Grüße 
Michael

bolliy, 2024-03-05:
modbus interface definitions:
https://github.com/bolliy/ioBroker.sun2000/blob/main/docs/definitions.md

Kaulii, 2024-03-05:
Super 👍 
Siehe hier:
[image]
die 15 sec sind was geschuldet?

Habe gerade den SL3000 noch auf das Update V300R023C10SPC311 gezogen, das war nicht aktuell. Wundert mich das die App das bisher nie bemängelt hat. Daher konnte ich auch die oben geschriebnenen Register nicht abrufen.
unter dem Register 40007 bekomme ich aber auch keine Uhrzeit ausgelesen. Sprich Wert 0
Siehe hier:
[image]

[image]
 Ich denke aber soweit funktioniert alles bestens!
Danke :-)🥇

Kaulii, 2024-03-05:
sun2000.0.inverter.1.info.model =SUN2000-20KTL-M2SUN2000-12
sun2000.0.inverter.1.info.modelID = 435

bolliy, 2024-03-05:
Hallo Michael,
wenn du weitere Datenpunkte von dem SmartLogger haben möchtest, kannst du mir diese gerne wieder per Email schicken.

Ich würde dich gerne verpflichten ein kleine Dokumentation (1 html Seite sollten reichen) in englisch zur Integration des Smartloggers zu schreiben, die ich gerne hier https://github.com/bolliy/ioBroker.sun2000/tree/main/docs einbinden würde. 
Den Inhalt der Doku als Fragen formuliert:
* Warum ist der SmartLogger sinnvoll - Funktionsbeschreibung (Aufzählung + Links)
* Vorraussetzungen müssen erfüllt sein. (Softwareversion, modbusID des Meters)
* usw.

Würdest du das für das mich machen?
LG Stepan

bolliy, 2024-03-05:
https://github.com/bolliy/ioBroker.sun2000/blob/68db6267fbf3645f053302eb90e46f7ad23c087d/lib/drivers/driver_inverter.js#L38

Hier siehst du, dass ich deinen WR nicht als Serie M1 einstufe. Das achaue ich mir gerne nochmal die https://www.photovoltaikforum.com/core/attachment/315028-solar-inverter-modbus-interface-definitions-pdf/ an. 
Ich melde mich ...
Stephan

bolliy, 2024-03-05:
die 15 sec sind was geschuldet? Eigentlich 5 Sec je Device.

Das werde ich noch für den SmartLogger anpassen, da der sicherlich robuster als der SDongle sien wird.

Kaulii, 2024-03-05:
Ah okay, die Geräte SUN2000-8KTL-M2, SUN2000-10KTL-M2, SUN2000-12KTL-M2, SUN2000-15KTL-M2, SUN2000-17KTL-M2 und SUN2000-20KTL-M2 können auch die Optimierer.

= 430 bis 435.

Mit der Dokumentation kann ich mich gerne Versuchen ;-) wird aber einige Tage dauern.

bolliy, 2024-03-06:
so Michael,
die M2 und höher werden nun auch wie M1 beahndelt. Also 2te BatteryUnit und Optimizer.

Intervall von 5 Sec gegenüber dem SmartLogger sollten nun auch möglich sein. 
Bitte testen und mir sagen ob es noch Probleme gibt. Und wie sieht das Timing aus? Also treten bei 5 Sec Intervall timeout im Log auf.
Wenn das gut läuft würde ich den SLogger in die v0.5.0 aufnehmen und publishen ;)
LG Stephan

Kaulii, 2024-03-06:
Hi Stephan,
werde ich gleich Testen!

aufpassen! an den M2 Geräten ist keine BatteryUnit mögich! Jedoch die Optimizer.
Aktuell gibt es eine neue Serie: die MB0 Geräte
Diese können dann auch BatteryUnit.

Kaulii, 2024-03-06:
Optimizer sind wieder im Register enthalten:👍 
[image]

Habe keine timeouts im log bei den 5sec.
[image]

läuft Perfekt!
Gebe nachmittags nochmal Rückinfo ob im Log ggf. Fehler sind.

LG Michael

Kaulii, 2024-03-06:
also an den M2 Geräten kann man keine Luna2000 anschließen. Ich hoffe wir reden jetzt vom gleichen :-)
Ja, darfst du schließen.

Läuft einwandfrei, im Log ist nichts!

bolliy, 2024-03-06:
Das der M2 keine Luna anschließen lässt, wusste ich nicht. Kannst du mir dann bitte den Battery Path **sun2000.x.inverter.0.battery** mal zeigen? Nicht, dass dort "unsaubere" States stehen.

Kaulii, 2024-03-06:
[image]
Alles Prima, der Ordner Battery ist leer.

Kaulii, 2024-03-06:
und hier vom 10 ktl
[image]

closed 2024-03-06 by bolliy