bolliy/ioBroker.sun2000 #42: Daten vom SCharger22

luckyheiko, 2024-01-30:
Halllo 

kein Bug nur ne Bitte, (frage)
ist es möglich auch die Daten von der WB anzeigen zu lassen?
die WB ist ja mit dem WR in 'Kontakt'.

in der Anleitung steht ja das Modbus aktive sein muss, damit die WB 'richtig' funktionert. 

Falls es möglch ist, danke schon mal 

mfg heiko

**Describe the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: <adapter-version>
 - JS-Controller version: <js-controller-version> 
 - Node version: <node-version> 
 - Operating system: <os-name>

**Additional context**  
Add any other context about the problem here.

bolliy, 2024-01-30:
Hallo Heiko, 
ich habe nur diese Information gefunden:
https://www.photovoltaikforum.com/thread/208282-modbus-register-f%C3%BCr-huawei-fusioncharger-wallbox/

In der Schnittstellenbeschreibung find ich leider nichts.
Stephan

bolliy, 2024-01-30:
Danke!
Kannst du deine WB auf dem Fusion Solar Portal sehen. Dann könnte die WB über eine modbus ID über den dongle angesprochen werden. Dann müsste dei modbus ID auch auf dem Fusion Solar Portal stehen. 
Ansonsten gibt es nur wenige Daten zum abgreifen. 
Da modbus eine serielle Verbindung ist, könnte die Integration im sun2000 Adapter erfolgen. Falls die WB über einen eigene ip adresse abgefragt werden muss, wäre das eher eine Aufgabe für einen neuen Adapter.
Kannst du mal nachschauen, wie die WB auf dem Portal eingebunden ist?

luckyheiko, 2024-01-30:
[image]
Ja, ist im Portal. 
und ja in der bedinungsanleitung steht drinnen, das für die Funktion der WB das TCP aktiviert sein muss. 
hab aber noch keine Info in der WB gefunden um der WB eine ID zu geben, aber ich denke (hoffe) das die Daten aus dem WR gelesen werden können.

geht ja nur um z.B. Ladeleistung und ob gerade eine Ladevorgang gestartet wurde.

bolliy, 2024-01-30:
FusionSolar/Gerätemanger/[Gerät klick]/Konfiguration

steht die **Modbus-ID-Adresse**

Hier als Beispiel mein Dongle:
[image]

luckyheiko, 2024-02-01:
gibt keine ID (hatte ich ja schon mal geschrieben)
[image]
und kann an der WB auch nichts einstellen
finde auch in der App keine ID oder sonstiges

bolliy, 2024-02-03:
Hallo Heiko,
das Thema ist doch umfangreicher als ich anfänglich gedacht habe. Siehe https://github.com/wlcrs/huawei_solar/issues/491

Da die Wallbox wahrscheinlich nicht über den Dongle mit einer eigenen modbus ID angesprochen werden kann, wäre das eher ein Thema für einen eigenen ioBroker Adapter. 
Ein issue könntest du hier https://github.com/ioBroker/AdapterRequests/issues einstellen.
LG Stephan

bolliy, 2024-02-11:
Hallo Heiko,
da ich nun ein Treiber-Model in den Adapter einbauen konnte, könnte ich mir vorstellen den Charger auszulesen.
Es sind ja tatsächlich nicht viele Daten.

Meine Idee: Ich erweitere den Adapter so, da er unterschiedliche Geräteklassen ansprechen kann. Dann müsstest du eine 2te Instanz des Adapters sun2000 installieren. Dort wird dann die ip und die modbusId des Chargers eingegeben.

Dafür benötige ich aber ausgiebige Unterstützung, da ich keinen Huawei Charger habe.
LG Stephan

luckyheiko, 2024-02-12:
klar kann ich dich hier versuchen zu unterstützen.
kann nur nicht immer 'sofort' antworten, da ich das Auto nicht immer zu Hause habe und auch 'meistens' nicht zu Hause lade. 
aber für Test oder so kann ich das Ding gerne mal anschließen.

(0.30er istaliert und bis jetzt scheint alles zu klappen)

bolliy, 2024-02-12:
Dann werd ich den Charger mal angehen. Kann allerdings etwas dauern. Melde mich sofern etwas zu testen gibt.
LG Stephan

arnovy, 2024-02-13:
Hi Stephan,

ich könnte auch meine Hilfe anbieten. Ich habe einen S-Charger 22 mit der V100R023C10SPC020

bolliy, 2024-02-15:
So ich habe eine **Entwickler Version** fertig gestellt ;)

Die Installation erfolgt über den **Expertenmodus**. Danach auf die „Krakenkatze“ klicken und dann die benutzerdefinierte Url
https://github.com/bolliy/ioBroker.sun2000/tarball/dev-charger
eingeben und die Installation starten.
Nach der Installation muss eine **neue Instanz** für den charger erstellt werden und die Intanz 0 **neu gestartet** werden!
In den Einstellunge der neuen Instanz (1) bitte die **Device Class** auf "charger" wechseln und die ip der WB und die modbusID eingeben.  

Ob das alles funktioniert - kann ich nich sagen. Ich habe halt keinen Huawei SCharger. 

Zu einem anderen Projekt habe ich gelesen, dass der SCharger noch gar nicht softwareseitig für die modbus Abfrage vorbereitet sein soll. Also unbedingt die aktuelle Software installieren.

https://github.com/evcc-io/evcc/discussions/8985

Lg Stephan

arnovy, 2024-02-15:
Hi Stephan,das werde ich gleich morgen testen saubere Arbeit und danke für die Entwicklerversion!Ich kenne das Projekt, da hatte ich geschrieben lt. release notes soll der 3rd Party Zugriff mit der altuellsten FW funktionieren. Was mir nicht ganz klar ist ob die Wallbox selbst eine Modbus Verbindung akzeptiert oder das auch über den Dongle laufen muss mit z.B. Modbus ID 255. Aber das werde ich morgen herausfinden.Wegen dieser grundsätzlichen Modbusdiakussion bin ich bereits seit geschlagenen 2 Wochen im Frage Antwortspielchen mit dem TAC Engineers von Huawei.. letzte Rückmeldung nach der kronkreten Nachfrage was denn nun z.B. im Modbus Adapter vom IObroker einzustellen ist hat damit geendet dass er die Frage selbst weitergeben muss. Das ganze gestaltet sich äußerst zäh.Ich bin gespannt auf den morgigen Test und melde mich!GrußAlex

https://github.com/bolliy/ioBroker.sun2000/tarball/dev-charger
evcc-io/evcc#8985

arnovy, 2024-02-16:
Hallo Stephan,

hab ich so installiert, als ziel IP habe ich die des S-Chargers genommen, Modbus Port 502, Modbus ID 1 mit folgendem Ergebnis:

[image]

ich habe auch Modbus ID 2 und 255 getestet ebenfalls mit gleichem Ergebnis.

Ich habe auch die IP des Wechselrichters mit Modbus Port 502 und Modbus ID 1,2 und 255 getestet mit folgendem Ergebnis:

[image]

folgende Objekte habe ich bzw. habe ich nicht:

[image]

arnovy, 2024-02-16:
ich habe noch eine Anfrage an das Huawei TAC offen mit der generellen Frage welche IP (ob inverter oder S-Charger) und welche Modbus ID für den „Modbus Adapter“ im IObroker für eine erfolgreiche Verbindung benötigt werden. Der Huawei Engineer musste die Anfrage weitergeben, jetzt warte ich auf Antwort. Das scheint ja weniger ein Problem der Register als ein generelles Connection Problem zu sein.. 

Das Thema auf EVCC kenne ich, dort hatte ich geschrieben, dass lt. den Releasenotes der neuesten FW 3rd Party Connection möglich sein sollen. 

Mal sehen was Huawei antwortet

bolliy, 2024-02-16:
Guten Morgen,
zumindest scheint dein SCharger eine Verbindung auf Port 502 (Bild 1) zu erlauben.  Versuch doch mal die modbusID **0**.
Ich glaube nicht, dass der SCharger über den Dongle gemappt wird. Der Dongle kann nur Inverter einbinden.

Ist der SCharger über das Fusion Solar Portal sichtbar. Vielleicht kann man dort die modbus Infos abrufen? 
Ist modbus tcp im Charger überhaupt aktiviert?

arnovy, 2024-02-16:
Modbus 0 mit gleichem Ergebnis: 
[image]

im Charger hatte ich für die Tests noch den IOBroker als ModBus Server angegeben:

[image]
[image]

arnovy, 2024-02-16:
Hier noch die Modbus Logs vom S-Charger vom ersten Verbindungsversuch direkt auf die Charger IP mit Modbus ID 1:

`024-02-16 09:10:41 EVT M801 tcpserver_modbus.cpp     [418]:one client 10.168.178.150:33698 connect server 10.168.178.48:502, mode:warning<NO SSL>, 1
2024-02-16 09:11:22 EVT M802 tcpclient_modbus.cpp     [1129]:Q Cnt:0x2,set equipaddr:0x64 2024-02-16 09:06:27
2024-02-16 09:11:22 ERR M801 tcp_modbus_base.cpp      [2113]:0001->0321 type:0635 para:5092 conseq:16 sta:3
2024-02-16 09:11:26 EVT M801 tcpserver_modbus.cpp     [418]:one client 10.168.178.150:33786 connect server 10.168.178.48:502, mode:warning<NO SSL>, 1
2024-02-16 09:11:41 EVT M801 tcpserver_modbus.cpp     [418]:one client 10.168.178.150:33810 connect server 10.168.178.48:502, mode:warning<NO SSL>, 1
2024-02-16 09:11:47 EVT M801 tcpserver_modbus.cpp     [795]:Q Cnt:0x5,usrTimeout 0 2024-02-16 09:06:56
2024-02-16 09:11:47 EVT M802 tcp_modbus_base.cpp      [2993]:conSta:5->6,S10.168.178.140:502,C10.168.178.48,8446
2024-02-16 09:11:47 EVT M801 tcpserver_modbus.cpp     [863]:Q Cnt:0x5,reg accept 0.0.0.0 2024-02-16 09:06:56
2024-02-16 09:11:47 EVT M802 tcp_modbus_base.cpp      [2997]:LinkDown timeout S:10.168.178.140:502,C:10.168.178.48,para:8446
2024-02-16 09:11:47 EVT M801 tcpserver_modbus.cpp     [496]:Q Cnt:0x2,tcp_modbus listen :0.0.0.1:502, ret 0 2024-02-16 09:06:56
2024-02-16 09:11:47 EVT M802 tcpclient_modbus.cpp     [438]:conSta:4->5,S10.168.178.140:502,C10.168.178.48,8446
2024-02-16 09:11:47 EVT M802 tcpclient_modbus.cpp     [1129]:set equipaddr:0x64
2024-02-16 09:11:57 EVT M801 tcpserver_modbus.cpp     [496]:T Cnt:0x2,tcp_modbus listen :10.168.178.48:502, ret 0 2024-02-16 09:06:57
2024-02-16 09:11:57 EVT M801 tcpserver_modbus.cpp     [887]:unreg mod 0x1016, para 0x43, server 10.168.178.48:502 para:0x844e
2024-02-16 09:11:57 EVT M801 tcpserver_modbus.cpp     [793]:reg mod 0x1016, para 0x4c, listen 0.0.0.1:502 para:0x844e 0x80, client num 1 mssLen 0 retry 10 exptTime 60
2024-02-16 09:11:57 EVT M801 tcpserver_modbus.cpp     [795]:usrTimeout 0
2024-02-16 09:11:57 ERR M801 tcp_modbus_base.cpp      [2113]:Q Cnt:0x1,0001->0321 type:0635 para:5092 conseq:12 sta:3 2024-02-16 09:08:04
2024-02-16 09:11:57 EVT M801 tcpserver_modbus.cpp     [863]:reg accept 0.0.0.0
2024-02-16 09:11:57 EVT M801 tcpserver_modbus.cpp     [496]:tcp_modbus listen :0.0.0.1:502, ret 0
2024-02-16 09:11:58 EVT M801 tcpserver_modbus.cpp     [887]:unreg mod 0x1016, para 0x4c, server 0.0.0.1:502 para:0x844e
2024-02-16 09:11:58 EVT M801 tcpserver_modbus.cpp     [793]:reg mod 0x1016, para 0x4d, listen 10.168.178.48:502 para:0x844e 0x80, client num 1 mssLen 0 retry 10 exptTime 60
2024-02-16 09:11:58 EVT M801 tcpserver_modbus.cpp     [496]:tcp_modbus listen :10.168.178.48:502, ret 0
2024-02-16 09:12:06 EVT M801 tcpserver_modbus.cpp     [418]:one client 10.168.178.150:33864 connect server 10.168.178.48:502, mode:warning<NO SSL>, 1
2024-02-16 09:12:45 EVT M801 tcpserver_modbus.cpp     [418]:one client 10.168.178.150:33912 connect server 10.168.178.48:502, mode:warning<NO SSL>, 1
2024-02-16 09:13:26 EVT M801 tcpserver_modbus.cpp     [887]:unreg mod 0x1016, para 0x4d, server 10.168.178.48:502 para:0x844e
2024-02-16 09:13:26 EVT M801 tcpserver_modbus.cpp     [793]:reg mod 0x1016, para 0x54, listen 0.0.0.1:502 para:0x844e 0x80, client num 1 mssLen 0 retry 10 exptTime 60 `

bolliy, 2024-02-16:
Die Verbindung kommt zustande - soweit ok. Ich habe mal die größe des Datenpakets verkleinert.
Versuche es bitte nochmal - also neue Version installieren ...
Stephan

luckyheiko, 2024-02-16:
hab gerade Instaliert und bekomme auch die 'fehler'
<html><body>

sun2000.1 | 2024-02-16 16:35:49.945 | info | Connected Modbus TCP to 192.168.10.19:502
-- | -- | -- | --
sun2000.1 | 2024-02-16 16:35:40.942 | info | Open Connection...
sun2000.1 | 2024-02-16 16:35:40.942 | info | Interval 40.894 sec
sun2000.1 | 2024-02-16 16:35:40.940 | warn | Error while reading from 192.168.10.19 [Reg: 4096, Len: 14, modbusID: 1] with: Timed out
sun2000.1 | 2024-02-16 16:35:08.992 | info | Connected Modbus TCP to 192.168.10.19:502
sun2000.1 | 2024-02-16 16:35:00.049 | info | Open Connection...
sun2000.1 | 2024-02-16 16:35:00.048 | info | Interval 39.113 sec
sun2000.1 | 2024-02-16 16:35:00.047 | warn | Error while reading from 192.168.10.19 [Reg: 4096, Len: 14, modbusID: 1] with: Timed out
sun2000.1 | 2024-02-16 16:34:29.464 | info | Connected Modbus TCP to 192.168.10.19:502
sun2000.1 | 2024-02-16 16:34:20.935 | info | Open Connection...
sun2000.1 | 2024-02-16 16:34:20.935 | info | Interval 32.148 sec
sun2000.1 | 2024-02-16 16:34:20.933 | warn | Error while reading from 192.168.10.19 [Reg: 4096, Len: 14, modbusID: 1] with: Timed out
sun2000.1 | 2024-02-16 16:33:55.822 | info | Connected Modbus TCP to 192.168.10.19:502
sun2000.1 | 2024-02-16 16:33:48.788 | info | Open Connection...
sun2000.1 | 2024-02-16 16:33:48.787 | info | Interval 38.444 sec

</body>
</html>

bolliy, 2024-02-16:
Sind eure Wallboxen mit dem vituellen Stromzähler verbunden? Somit wäre eine modbus Abfrage nicht möglich.
Siehe https://solaranzeige.de/phpBB3/viewtopic.php?t=4333

luckyheiko, 2024-02-16:
hmpf, 

ok, aber danke für die Info 
und ja, ich habe (wahrscheinlich jeder) hat diesen virtuellen zähler

bolliy, 2024-02-16:
Da die Überschußladefunktion von Huawei eh nicht gut funktionieren soll, könnte man diese natürlich ersetzen.
Hängt doch mal den SCharger aus dem System und testet ob es dann funktioniert....

bolliy, 2024-02-16:
Und es gibt noch ein Problem. Da der SCharger über modbus auf den WR zugreift, bekommen wir beim Auslesen der Daten vom WR timeout-Fehler. Da ein konkurierenden Zugriff über modbus nicht funktioniert. Das Setup ist also keine gute Lösung!

luckyheiko, 2024-02-16:
ich kanns leider nicht ändern, da mein Zählerkasten eh schon aus allen näten platzt
könnte ich hier nicht noch einen  Smartmeter einbauen :(
von daher muss ich leider damit leben (hab keine lust auf 2000+€zählerkasten umbau)

bolliy, 2024-02-16:
Ich habe eine Wallbe Wallbox auch ohne Zähler! Die habe ich auch per modbus angebunden - funktioniert super. Habe  dort noch nie ein timeout gesehen. Ich berechne die Leistung anhand der angeschlossenen Phasen*A*220V*Powerfaktor und steuere so den Überschuß in mein Leaf. Den Verbrauch der WB ermittlere ich also nur indirekt. Das wäre auch eine Lösung.

arnovy, 2024-02-16:
Danke für eure Erkenntnisse! Sollte der FE Zähler mal erschwinglicher werden werde ich es mir überlegen. Aber >200€ sind mir zu viel für den Zweck der Modbusabfrage.
Leider kann man den virtuellen Zähler auch nicht ohne weiteres löschen und wenn doch verbindet sich die Wallbox automatisch wieder damit.

arnovy, 2024-02-17:
Eine Idee noch: Wie kommen denn die Werte der Wallbox in die Fusion Cloud? Könnte man nicht hier ansetzen und diese Werte auslesen?

bolliy, 2024-02-17:
Es gibt ein Adapter für die Huawei Fusion Solar Cloud: https://github.com/KornSW/ioBroker.fusionsolar

Erstmal vielen Dank für eure Unterstützung! 
Wie oben ausgiebig diskutiert macht es zur Zeit keine Sinn an der SCharger Integration weiterzuarbeiten. 
Deshalb werde ich diesen issue auf onHold + wontfix setzen aber nicht schließen. Wenn es sich neue Erkenntnisse ergeben sollten, können wir gerne an dieser Stelle weitermachen.

Stephan

bolliy, 2024-02-25:
Neue Test-Version ist raus! Nun könnt ihr eure SCharger über den integrierten modbus-proxy einbinden.  

Siehe: https://forum.iobroker.net/post/1132998

arnovy, 2024-02-25:
Hi Stephan,

der Modbus Proxy macht Probleme:

[image]

Ich habe den IoBroker als Docker laufen, erst dachte ich es könnte in Rechteproblem sein, aber die Ausführung des Containers als root bringt die gleiche Fehlermeldung

bolliy, 2024-02-25:
google mal nach: docker ignoring ip-address service will listen on '0.0.0.0'

arnovy, 2024-02-25:
Hab’s auch mit der IP des Dockers selbst versucht gleiches Problem, allerdings hab ich die Einstellung noch nicht als Root getestet

bolliy, 2024-02-25:
Ich habe es über virtualBox und auf einem Raspi laufen. 
Leider kann ich den Exception (noch) nicht abfangen, da das modul modbus-serial noch ein Bug hat - ist auch gefixt aber noch nicht released.

arnovy, 2024-02-25:
Gleicher Fehler auch wenn ich den Docker als Root ausführe und die Docker IP beim Modbus Proxy angebe

bolliy, 2024-02-25:
Docker muss richtig konfiguriert werden! Kann dir allerdings nicht den genauen Aufruf nennen.

arnovy, 2024-02-26:
okay, ich habe das Portproblem mit einer Portweiterleitung in IPtables umgangen:
sudo iptables -I INPUT 1 -p tcp --dport 502 -j ACCEPT
sudo iptables -A PREROUTING -t nat -i eth0 -p tcp --dport 502 -j REDIRECT --to-port 5022

ich leite den Port 502 damit auf Port 5022 in den Docker um.
Es scheint irgendein Rechteproblem mit wellknown ports <1024 zu geben.

Allerdings habe ich bereits das nächste Problem entdeckt:

[image]

ich habe Modbus 100 auch mal auf Modbus 1 geändert mit dem gleichen Ergebnis.
Was mir nicht ganz klar ist, der S-DongleA in der Modbus Proxy konfig ist das die ModbusID die der Modbus Proxy bereitstellt oder handelt es dich hierbei um die ID die der Adapter beim Wechselrichter Abfragt?
Warum unterschiedest du ModbusID1 und SDongelA ModbusID 100? Der Wechselrichter wird doch standardmäßig per ModbusID 1 abgefragt

bolliy, 2024-02-26:
der SDongle hat eine eigene UnitID (modbusID). Die ist eigentlich immer 100. Die Daten des SDongle sind wichtig für die Wallbox!
Der erste Inverter ist meist die modbusID 1 usw.
Der SDongle hält die einige summierte Werte (über alle Inverter) bereit.

bolliy, 2024-02-26:
SdongleA modbus interface definition (Issue 2, 2023-04-20): https://photomate.zendesk.com/hc/en-gb/articles/7275970817437-SDongleA-MODBUS-Interface-Definitions

bolliy, 2024-02-26:
Würdest die deine Fragen/Erfahrungen/Lösungen bitte im Forum teilen. Damit andere User auch davon profetieren können.
https://forum.iobroker.net/topic/71768/test-adapter-sun2000-v0-1-x-huawei-wechselrichter

Danke!

arnovy, 2024-02-26:
nach Abstecken/Stecken des Dongles hat sich der Adapter mit dem S-Dongle verbunden.
Problem jetzt: Ich bekomme die Wallbox nicht dazu den Stromzähler auf den ModBus zu ändern.
Den Virtuellen Stromzähler habe ich gelöscht bekommen, indem ich den S-Dongle abgesteckt habe, anschließend konnte ich den Virtuelle Zähler löschen. Allerdings ist es mir nicht gelungen den Modbus Proxy einzubinden.
Ich denke das Problem hier ist, sobald der Dongle abgesteckt ist kann ich zwar die IP des Proxys manuell eingeben, hinzufügen lässt er sich aber nicht, da ja der Dongle ausgesteckt ist. Sobald ich aber den Dongle wieder einstecke fügt die Wallbox diesen automatisch wieder als virtuellen Zähler hinzu und ich kann den ModbusProxy nicht mehr hinzufügen weil er schon einen Zähler hat. Hier dreht man sich im Kreis

bolliy, 2024-02-26:
Da ich keinen SCharger von Huawei habe, kann ich das natürlich nicht richtig testen. Aber werden andere Daten vom SDongle gelesen oder geschrieben. Im Log kannst du aber sehen welche Daten der SCharger abfragt. Es ist im JSON-Format aufgebaut und sieht ähnlich dem hier aus:
[image]

Ich bräuchte dann die Daten als JSON:
Modbus tcp server: {"stat":{"#getMultipleHoldingRegisters-address_32064-value_2-unidId_1":1,"#getMultipleHoldingRegisters-address_37765-value_2-unidId_1":4,"#getMultipleHoldingRegisters-address_37100-value_38-unidId_1":1,"#getMultipleHoldingRegisters-address_32080-value_2-unidId_1":1,"#getMultipleHoldingRegisters-address_37758-value_30-unidId_1":4,"getMultipleHoldingRegisters-address_37498-value_20-unidId_100":5168,"getMultipleHoldingRegisters-address_32080-value_2-unidId_1":6229,"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":6228,"getMultipleHoldingRegisters-address_37100-value_38-unidId_1":5157,"#getMultipleHoldingRegisters-address_37000-value_50-unidId_1":1,"getMultipleHoldingRegisters-address_37765-value_2-unidId_1":6234,"#getMultipleHoldingRegisters-address_47081-value_18-unidId_1":1,"getMultipleHoldingRegisters-address_37758-value_30-unidId_1":6225,"getMultipleHoldingRegisters-address_37000-value_50-unidId_1":1326,"getMultipleHoldingRegisters-address_47081-value_18-unidId_1":1324,"getMultipleHoldingRegisters-address_32000-value_116-unidId_1":1322}}

Hier kann man gut sehen, was über modbus-proxy so läuft. Die Einträge mit # werden (z.Zt.) nicht bedient.

bolliy, 2024-02-26:
Vielleicht bringt dich dieses weiter?

https://tff-forum.de/t/pv-ueberschussladen-smart-charger-huawei/290268?page=2

luckyheiko, 2024-08-19:
hab das irgendwie noch nicht wirklich rausbekommen.
deswegen hier noch mal kurzne 'doofe' frage.

es ist ja der Proxy nun in der Instanz. Hier hatte ich gedacht, das nun die WB mit ausgelesen werden kann.
wenn ich diesen Aktiviere (wegen WB) finde ich aber keine Werte im IObroker, kann man dann irgendwas aus der WB auslesen?
oder ist beim auslesen der WB gar nichts möglich?
v0.11.0
geräte IP x.x.44.99
wb IP x.x.44.95
ID WR 1
ID Dongle 100

bolliy, 2024-08-19:
@luckyheiko über den modbus-proxy kannst du den SCharger leider nicht einbinden. Der SChrager muss direkt mit dem WR kommunizieren. Das wird erst mit der Integration der Emma möglich sein. Da bin ich allerdings noch nicht dran, da ich keine Emma habe.

LG Stephan

luckyheiko, 2024-08-20:
danke für die info.
schönen tag noch

bolliy, 2024-09-29:
Der SCharger wird standardmäßig mit dem virtuellen Stromzähler des Wechselrichters per modbus verbunden. Dann kann man allerdings nicht mehr auf den SCharger per modbus zugreifen. 
Als Lösung sehe ich nur die Verwendung der Emma von Huawei.

closed 2024-09-29 by bolliy