bolliy/ioBroker.sun2000 #44: Weitere Zustände berechnen

Obsthaendler, 2024-02-02:
Es wäre hilfreich wenn durch den Adapter automatisch weitere Zustände berechnet werden, z.B. Grid, dieser wird berechnet vom InputPower (loss?!) - HouseConsumption. Diesen Wert benötigt u.a. gerne EVCC.io

Des weiteren wären weitere Werte wie beim Huawei Portal gut für eine Auswertung:
- Autharky in %
- Eigenverbrauch in %

Danke für die Entwicklung!!!

Gruß

bolliy, 2024-02-03:
Die **Netzeinspeisung** findest du bereits im State:
sun2000.0.meter.activePower (>0: feed-in to grid. <0: supply from grid.)

**Autharkiegrad** und **Eigenverbrauch** können schon jetzt aus anderen States abgeleitet werden. Hier sind die Zusammenhänge ganz gut dargestellt.

Da ich gerade mit der Umstrukturierung des Adapter befasst bin, werde ich mich vorest nicht mit der Umsetzung befassen.

LG Stephan

bolliy, 2024-03-05:
Ich habe in der Doku zum Adapter aufgezeigt wie du die Werte selbst berechnen kannst. Umsetzen kannst du das mit etwas js Zeilen oder blockly.

https://github.com/bolliy/ioBroker.sun2000/blob/main/docs/calculation.md

Stephan

Obsthaendler, 2024-03-05:
Sehr gut. Dann kannst du das hier schließen.

closed 2024-03-05 by Obsthaendler