bolliy/ioBroker.sun2000 #60: Datenabfrage nur am Tag

Skol6, 2024-03-16:
### What would you like to be added?

Hi,

mein Sun2000 Wechselrichter ist nur am Tag aktiv, wenn die Module entsprechend Strom erzeugen.
Sonst deaktiviert er sich komplett. Das lässt sich leider auch nicht anders einstellen.

Das bedeutet das die Instanz in der Nacht Unmengen an Fehler produziert, da sie ja mit versucht Kontakt aufzunehmen.
Wäre das als Erweiterung möglich das irgendwie einzustellen?

Dankeschön!

### Why is this needed?

Wechelsrichter nur am Tag aktiv, sonst nur Fehlermeldungen weil natürlich keine Datenverbindung verfügbar.

bolliy, 2024-03-16:
Mit der v0.5.1 müsste das eigentlich funktionieren. Siehe im Wiki hier: https://github.com/bolliy/ioBroker.sun2000/wiki/Standby-Detektierung.

Hier https://github.com/bolliy/ioBroker.sun2000/wiki/Adapter-installieren wird die Installation über npm beschrieben.
LG Stephan

Skol6, 2024-03-16:
Danke für die Info Stephan.
Hab das Update gleich aufgespielt. Mal schauen was die Sonne morgen sagt!
bg

bolliy, 2024-03-16:
Ich freue mich auf dein Feedback.

Skol6, 2024-03-16:
mhmm also bei mir spuckt er noch Fehlermeldungen aus
```
sun2000.0 | 2024-03-16 21:24:01.174 | warn | Couldnt connect Modbus TCP to 192.168.1.26:502 connect EHOSTUNREACH 192.168.1.26:502
sun2000.0 | 2024-03-16 21:23:58.170 | info | Open Connection...
sun2000.0 | 2024-03-16 21:23:58.169 | info | Interval 52.024 sec
sun2000.0 | 2024-03-16 21:23:58.169 | warn | Error while reading from 192.168.1.26 [Reg: 30015, Len: 56, modbusID: 100] with: connect EHOSTUNREACH 192.168.1.26:502
sun2000.0 | 2024-03-16 21:23:58.169 | warn | Adjustment: It has difficulty calibrating. The current step is 0
sun2000.0 | 2024-03-16 21:23:58.168 | warn | Couldnt connect Modbus TCP to 192.168.1.26:502 connect EHOSTUNREACH 192.168.1.26:502
sun2000.0 | 2024-03-16 21:23:55.162 | info | Open Connection...
```

bolliy, 2024-03-16:
Der Adapter benötigt natürlich erstmal eine Verbindung zum WR. Erst dann kann er den stanby überhaupt detektieren. Die standby-Detektierung erfolgt über ein Daten aus dem WR. Henne-Ei-Problem ;)
Also morgen Abend abwarten. 
Stephan

bolliy, 2024-03-16:
Nach Sonnenaufgang die **auto-adjust** Funktion aktivieren, damit der Adapter die richtige modbus-timing Daten ermitteln kann.
https://github.com/bolliy/ioBroker.sun2000/wiki/auto%E2%80%90adjustment

Skol6, 2024-03-17:
Leider kommen nach Sonnenuntergang wieder die Fehlermeldungen.
Irgendwie klappt die Standby Detektierung des Adapters noch nicht

bolliy, 2024-03-17:
ok,
was steht im State sun2000.0.inverter.0.deviceStatus ?
Der Adapter erkennt bei dem Wert = 2 (Hex 0x0002) den standby der WR. Zumindestens war das die Aussage eines anderen Anwenders. Mein WR hat eine Batterie und geht nicht in den stanby mode.

Ich brüchte auch die States:
sun2000.0.inverter.0.info.model
sun2000.0.inverter.0.info.modelID

LG Stephan

bolliy, 2024-03-18:
Ist der Längengrad und der Breitengrad im ioBroker System hinterlegt?

Skol6, 2024-03-18:
> ich habe gerade einen Fehler in der letzten Test-Version v0.5.1 festgestellt. Wird in der nächsten Test-Version gefixt.

Perfekt.

Einfach bescheid geben, dann teste ich es gerne.

bolliy, 2024-03-21:
Ich habe gerade v0.6,0 veröffentlicht: https://forum.iobroker.net/post/1142965

Bitte testen. Du brauchst aber Licht auf deinen Modulen, sonst findet der Adapter deinen WR nicht.

Skol6, 2024-03-21:
Danke. Habs schon aktualisiert. Mal schauen was morgen abend passiert!

Skol6, 2024-03-24:
Leider ist der Fehler noch immer da

bolliy, 2024-03-24:
ok, das ist ja ärgerlich.
Ist in den **Systemdaten** von iobroker **Längergrad** und **Breitengrad** hinterlegt? 
Ansonsten bitte eintragen. 

Kannst du mal im Log schauen, ob dort nach Sonnenuntergang folgende Meldung auftaucht:

"The Inverter 0 switches to Standby: detecting irradiation mode" 

steht?

Skol6, 2024-03-24:
Grade sind eingetragen.
Meldung taucht nicht in den Logfiles auf - Muss ich einen speziellen Log Modus aktivieren?

bolliy, 2024-03-24:
Nein, ist ein info-Log - taucht im normalerweise im Log auf.
Da ich kein WR habe, der ohne Speicher in den standby geht, werde ich ein Simulator bauen.
Also über den modbus-proxy auf einen Adapter zugreifen, der so tut als ob er in den Standby bei sonnuntergang geht. 

Das wird einige Tage dauern, da die Funktionsweise über nacht testen muss.
Ich melde mich wieder...
Lg Stephan

Kaulii, 2024-03-26:
Bei meinem20ktl-M**2** (hat die gleiche Firmware wie der M0) (dieser kann keinen Speicher) habe die States:
[image]
 aktuell ist der Wechselrichter im Standby.

-> habe keine Probleme
@Skol6 welche Firmware Version hat dein Wechselrichter?

bolliy, 2024-03-29:
Ich habe einiges umgebaut. 
Geht dein WR auf deviceStatus = 2 (Sonnenuntergang), wird nur noch der deviceStatus alle 60 Sekunden abgefragt. Die Fehlermeldung werden dann aber unterdrückt. Erst wenn der WR wieder aufwacht, erfolgen die normalen Abfragen. 
 
Um den WR zu erkennen, benötige der Adapter weiterhin tageslicht. Sonst geht der Adapter schlafen...

Bitte per [github ](https://github.com/bolliy/ioBroker.sun2000/wiki/Adapter-installieren) aktualisieren. Die Version bleibt auf v0.6.1.

Lg Stephan

Obsthaendler, 2024-03-30:
Es klappt zwar mit der Erkennung aber der Adapter startet trotzdem ständig neu. 

[image]

bolliy, 2024-03-30:
Nach dem Event **TERMINATE_YOUSELF** würde ich denken, dass der restart nicht vom Adapter kommt. 
Wenn der Adapter abstürzt, würde ich von sentry informiert. Hast du sentry vielleicht deaktiviert?

bolliy, 2024-03-30:
Ich versuche es gerade bei mir nachzubauen....bei mir startet der Adapter nicht eigenständig neu. Das ist recht eigenartig.
Hast du vielleicht ein script laufen, dass den Adapter neu startet?

bolliy, 2024-03-30:
Hast du vielleicht einen auto. Neustart hinterlegt?

[image]

Obsthaendler, 2024-03-31:
> Hast du vielleicht einen auto. Neustart hinterlegt?
> 
> ![grafik](https://private-user-images.githubusercontent.com/36469725/318221661-61d17d12-369f-4e96-ac1a-0347376eb5ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE4NjU2MzMsIm5iZiI6MTcxMTg2NTMzMywicGF0aCI6Ii8zNjQ2OTcyNS8zMTgyMjE2NjEtNjFkMTdkMTItMzY5Zi00ZTk2LWFjMWEtMDM0NzM3NmViNWVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzMxVDA2MDg1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZhNjQ5MmE3ZWI1Y2RmYTBiNjAyZmM1NTI2YTFkODQ5N2VhYzkwNWJiYTE1ZjI4ZjA3MmYxMmM1YjlmYTYwZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5w6ZPTqwk5kGeGLu1kNF77Vt7fdROpolvkffP8GUPwM)
> 
> Siehe Meldung: Terminated (START_IMMEDIATELY_AFTER_STOP): Without reason

So war es. Hatte meinen Kommentar eigentlich wieder gelöscht weil es mir dann selbst aufgefallen ist. So dämlich von mir.

bolliy, 2024-03-31:
Kein Problem. So habe ich auch etwas gelernt ;)

bolliy, 2024-03-31:
Zu Ostern wurde eine neue Testversion sun2000 Adapter v0.6.2 veröffentlicht.
https://forum.iobroker.net/post/1145740

Lg Stephan

Skol6, 2024-04-03:
Hi Stephan, 
mit der Version schauts jetzt super aus.
`sun2000.0 | 2024-04-03 18:58:35.415 | info | The inverter with modbus ID 1 is no longer accessible. That is why the logs are minimized.
-- | -- | -- | --
sun2000.0 | 2024-04-03 18:58:35.415 | info | The Inverter with modbus ID 1 switches to Standby: detecting irradiation mode.`
Ich glaub das Thema ist gelöst!
Vielen lieben Dank!

closed 2024-04-17 by bolliy