bolliy/ioBroker.sun2000 #61: Maximale Entladeleistung des Speichers (müsste Register 47077 sein) beschreibar machen

Dimi1305, 2024-03-18:
### What would you like to be added?

Hallo Stephan,

da du dich gerade mit dem Beschreiben der Register befasst würde ich gerne einen Wunsch äußern. Vielleicht hast du es auch schon in Planung.
Beim Laden des Autos über die Wallbox (go-e) würde ich gerne über modbus die Speicherentladung der Luna2000 auf 500 W begrenzen. Es macht für mich keinen Sinn den Speicher komplett in das Auto zu entladen. Momentan muss ich das über die fusionsolar Webseite einstellen. Dort kann man den Wert von 5000 W auf einen beliebigen Wert (0-5000) einstellen.

Danke für den tollen Adapter.
Dimi

### Why is this needed?

.

bolliy, 2024-03-19:
Hallo Dimi,
genau das geiche Problem möchte ich für meine Anwendung auch gelöst sehen. Bin gerade dabei für die nächste Testversion einige Beispielszenarien aufzustellen.
Die **Speicheentladebegrenzung** kommt auch noch ;)
Hier schon mal die ersten Sätze im Wiki. https://github.com/bolliy/ioBroker.sun2000/wiki/Battery-control
LG Stephan

Dimi1305, 2024-03-19:
Hallo Stephan,

du bist der Beste 👍
Ich freue mich auf die kommenden Feautures.

LG Dimi

arnovy, 2024-03-21:
Das würde mir unglaublich helfen, zumal der S-Charger immer mal wieder sinnfrei aus der Batterie zieht. Da die Modbus oder API Schnittstelle beim S-Charger ja immer noch nicht funktionieren wäre das ein gangbarer Weg das zu unterbinden

bolliy, 2024-03-21:
Ich habe gerade v0.60 veröffentlicht: https://forum.iobroker.net/post/1142965

arnovy, 2024-03-21:
Wow das ging schnell! Genau so brauch ich das!
Jetzt kann ich über einen Taster den ich vor dem Ladevorgang bediene die Batterie sperren.
Vielen Dank für deine hervorragende Arbeit!

bolliy, 2024-03-23:
Ich würde diesen issue gerne schließen. 
Ist das für euch ok?

arnovy, 2024-03-23:
Top! Geht wie gewünscht, ich habe heute mehrfach das EV geladen ohne die Batterie zu entladen!

Dimi1305, 2024-03-23:
Hallo Stephan,

also das mit der Entladebegrenzung funktioniert wie gewünscht. Die anderen Funktionen habe ich noch nicht getestet. Von meiner Seite aus kann der issie geschlossen werden.

LG Dimi

closed 2024-03-23 by bolliy