bolliy/ioBroker.sun2000 #63: EMMA Huawei A-02

gmihalca, 2024-03-22:
### What would you like to be added?

Hi, glaubst Du das auch die EMMA irgendwann integriert wird??

(https://ske-solar.com/produkt/huawei-energy-management-assistant-emma-a02/)

### Why is this needed?

https://ske-solar.com/produkt/huawei-energy-management-assistant-emma-a02/

bolliy, 2024-03-22:
Ich fürchte dir erstmal einen Korb geben zu müssen. Ich will das nicht komplett ausschließen! Nur zur Zeit braucht der Entwickler eine Auszeit ;) Zuerst möchte ich, dass die vorhandenen Funktionen ausreichend getestet werden. Danach ist eine neue stable Version fällig.
Sofern wir genügend Unterstützer finden und ich Zugang zur der Hardware bekomme, kann ich mir die Umsetzung vorstellen.
Lg Stephan

gmihalca, 2024-03-23:
Hi, und danke für die rasche Antwort.

Du hast wirklich eine TOP Arbeit bis jetzt mit dem Adapter geleistet und ich bin sehr dankbar dafür, hätte nur gehofft das ich mit dem EMMA es so einstellen kann das mein Akku nicht entladen wird wen ich mein Auto auflade den es nicht so produktiv ist, hast du irgendeine Idee was ich dagegen machen kann??

Hab den Logger300 im Einsatz.

Und ich würde gerne deine Arbeit über PayPal unterstützen, wenn möglich.

Lg
Gheorghe

bolliy, 2024-03-23:
Hallo Gheorghe,

über eine Spende würde ich mich natürlich freuen :-)
https://www.paypal.com/donate/?hosted_button_id=ZTX3VP9LZBDCG
oder
https://github.com/sponsors/bolliy

Ich habe keine modbus Schnittstellenbeschreibung für die Emma gefunden.

Hier einige Informationen aus anderen Projekten:
https://github.com/wlcrs/huawei_solar/discussions/457
https://github.com/wlcrs/huawei_solar/issues/491

Sofern du die modbus definition bekommst, kann ich mal schauen wie ich die Emma ansprechen kann.
Ein eigene js wäre denkbar, dass dann ggf. die gewünschte Einstellung vornimmt.

Lg Stephan

bolliy, 2024-03-23:
Danke Gheorghe für deine Spende!
Habe mich sehr gefreut. 

Stephan

bolliy, 2024-03-23:
Welche Adapterversion setzt du ein? Kann die neue **Battery control** Funktion nicht dein Problem lösen?
https://github.com/bolliy/ioBroker.sun2000/wiki/Battery-control

ist in der v0.6.0 implementiert. Kannst über npm installieren https://github.com/bolliy/ioBroker.sun2000/wiki/Adapter-installieren

Stephan

ismiregal17, 2024-05-27:
Hallo Stephan, ich wollte auch mal Danke sagen. Mega was deine Integration so alles vermag. 
Nur habe ich leider das Problem mit der EMMA, hab sie relativ früh bekommen und nun musste ich meine HA Integration über die Fusion Solar mit API machen. Das geht schon mal reicht aber bei weitem nicht an deine Integration heran. 
Ich hätte die Kommunikations-Matrix der aktuellen Software und die der kommenden Software. Sind Excel Datei’n. Wie kann ich die die Zukommen lassen?
Grüße

bolliy, 2024-05-27:
gerne an: stephan@mante.info
Stephan

bolliy, 2024-05-27:
@ismiregal17 sollen mich die Dokumente im Bezug auf die Integration der EMMA in den Adapter weiter bringen? Würdest du mir dein Anliegen bitte näher erläutern.
Stephan

ismiregal17, 2024-05-27:
@stephan, sehr gerne. Zum jetzigen Zeitpunkt, mit angeschlossener EMMA funktioniert deine Integration leider nicht mehr. Da es zu keinem Verbindungsaufbau über Modbus TCP kommt, somit können die Register nicht mehr ausgelesen werden.
Durch die FusionSolar Integration kann man auf  Inverter und Battery zugreifen nur da fehlt halt Netzbezug und - Einspeisung und die Optimierer Daten fehlen gänzlich. Die einzigen Daten die Abrufbar sind, sind die Standortdaten.

bolliy, 2024-05-27:
Hallo Fabian,
nach meinem Kenntnisstand ist es z.Zt. nicht möglich die Emma einzubinden. 
Siehe hier https://github.com/bolliy/ioBroker.sun2000/issues/63#issuecomment-2016468274

LG Stephan

bolliy, 2024-07-22:
Habe von Fabian die neue mobus Interface definition der Emma V100R024C00 bekommen:
https://www.photovoltaikforum.com/core/attachment/479611-smarthems-v100r024c00-modbus-interface-definitions-pdf/
Nun kann ich darüber nachdenken die Emma in den sun2000 Adapter zu integrieren...

Hierzu habe ich an euch ein paar Fragen:
1. Steht die modbus Schnittstelle der sun2000 Inverter noch zur Verfügung, wenn die Emma eingebunden wurde? Oder ist die modbus Schnittstelle durch die Emma belegt?
2. Wird der SDongle am WR noch vorhanden oder wurde dieser am Master entfernt?

Da kommen von mir bestimmt noch später weitere Verständnisfragen.

LG Stephan

gmihalca, 2024-07-22:
Hi, freut mich zu hören das es doch was werden könnte.

1. Also die modbus Schnittstelle der sun2000 ist durch Emma belegt Du must die Emma genau so wie ein Smartlogger betrachten.

2. Die SDongle wird nicht mehr benötigt, auch hier ist es genau wie beim Smartlogger  nur das es kein Smartmeter mehr benötigt wird den es ist ja die Emma.
[QuickGuide_Three-Phase_PVESSEMMA-Networking_V03_2024-01-31_EN.pdf](https://github.com/user-attachments/files/16339312/QuickGuide_Three-Phase_PVESSEMMA-Networking_V03_2024-01-31_EN.pdf)

lg
Gheorghe

bolliy, 2024-07-23:
Hallo Gheorghe,
wenn ich das richtig sehe sind die Informationen von dem sun2000 WR nur summiert über die Emma abrufbar. Im Gegensatz zum SmartLogger, der die Daten der WR 1:1 durchreicht. Die detaillierten modbus Registerinformationen können somit nicht dargestellt werden.
Daraus ergibt sich nur ein geringer Bezug zum sun2000 Adapter. Es müsste ein neuer Adapter für die Emma entwickelt werden, der auf dem sun2000 Adapter fußt.

Wie ist eure Einschätzung hierzu?

Lg Stephan

gmihalca, 2024-07-24:
Hallo Stephan,

Stimmt so nicht ganz, es zeigt schon gesammelt  aber das ist nur die Emma und es ist ja richtig so.

Warum ich sag das es wie beim Logger ist.

1. es spricht mit jedem WR bis 3 stk. über die ModBus Adresse die ja man eingeben könne  soll wie beim Logger3000

2. dann holt es sich es die Werte direkt von WR wie auch jetzt beim Logger

3. was es ja nicht hat ist ja ein Smartmeter weil es ja selbst einer ist und dafür hätte ich ja die Abfragen gesehen, somit sollte es ja sehr dem Logger ähneln 

Hoffe ein wenig geholfen zu haben.

Lg
Gheorghe

bolliy, 2024-07-24:
Hallo Gheorghe,
ja ähnlich dem SmartLogger - nur dieser reicht die Daten der einzelnen WR per modbus 1:1 weiter. Die Registeradressen sind identisch. Dieses ist aber bei der Emma nicht der Fall. So gibt es auch I64 und U64 Register, diese Größenordnung gibt es beim sun2000 nicht.

Da ich leider keine Emma habe, wird die Entwicklung schwierig für mich. Die Implementierung ist nur möglich wenn ich direkt per modbus auf die Emma zugreifen kann.
Könnte du/ihr dir/euch vorstellen, dass ich per vpn-tunnel oder tcp-forwarding auf eine Emma zugreifen könnte?

LG Stephan

gmihalca, 2024-07-25:
Hi, 

Okay verstehe.

Ich hab die Emma leider ausgebaut und verwende derzeit den Logger wieder eh deswegen.

Ich könnte die dir schicken und du machst mal Vorort bei dir die Versuche.

Sonst liegt die noch länger bei mir im Kasten.

bolliy, 2024-07-25:
Hallo Gheorghe,

das wäre natürlich richtig super. Kann ich die Emma auch betreiben ohne meine Installation komplett umzubauen?
Oder könntest du mir helfen die Emma bei mir einzubinden?

LG Stephan

gmihalca, 2024-07-25:
Hi Stephan,

naja ziemlich blöd ist es schon, denn es kann schon durchaus sein das du deine Werte aus der Fusionsolar verlieren kannst,
Und ja ein wenig Umbauarbeiten hast bestimmt, ich schick dir nochmal den Plan dazu, kannst damit nichts falsch machen.

Schick mir dann noch deine Adresse und ich schick dir die Emma zum Probieren.

Lg
Gheorghe

cksoft0807, 2024-07-26:
> Hallo Gheorghe,
> 
> das wäre natürlich richtig super. Kann ich die Emma auch betreiben ohne meine Installation komplett umzubauen? Oder könntest du mir helfen die Emma bei mir einzubinden?
> 
> LG Stephan

Mit EMMA muss die Anlage komplett neu angelegt werden. Also Geräte aus der Anlage entfernen und zur neuen Anlage mit der Emma hinzufügen. Smartmeter und Dongle sind dann über und zur Installation braucht es eigentlich eh auch einen Eli.

wolfganghuse, 2024-10-02:
Hallo, ich wäre auch an einer Emma Integration interessiert... Ich habe hier eine im Einsatz und würde für weitere Diagnose zur Verfügung stehen.

Gruß,
Wolfgang

bolliy, 2024-10-03:
Ich brauch mal eure Unterstützung.
Mir ist nicht ganz klar wie der/die Power Meter in der Emma organisiert sind. In der [Modbus Interface Definition](https://www.photovoltaikforum.com/core/attachment/479611-smarthems-v100r024c00-modbus-interface-definitions-pdf/) sind mehrer Datenbereiche im Meter Managment vorhanden:
1. built-in electric energy sensor, Seite 9-12
2. external electric energy sensor, Seite 12-15

Da ich kein Emma habe, versuche ich das Emma System zu verstehen.
Hat die Emma einen eingebauten Energy Meter oder muss immer ein externer energy sensor angeschlossen werden.
Welcher Datenbereich ist für die Meterdaten interessant?

LG Stephan

wolfganghuse, 2024-10-03:
Ich bekomme hoffentlich morgen die Modbus Schnittstelle freigeschaltet. 
Danach kann ich hoffentlich ein paar Infos liefern.

bolliy, 2024-10-03:
Wenn ich die Installationsanleitung richtig verstehe, wird im allgemeinen der eingebaute Energy Meter verwendet. Falls ich hier falsch liegen sollte - bitte meldet euch bitte schnell hier.
Ich habe nun mit der Projektierung angefangen und werde bald mit der Umsetzung anfangen…

LG Stephan

cksoft0807, 2024-10-03:
Die Emma ist Dongle und Smartmeter in einem Gerät. Es gibt allerdings interne Messung und externe Messklemmen am Kabel. Könnte möglicherweise auch das meinen.

bolliy, 2024-10-03:
Es gibt aber auch auf Seite 12 wird die "Register Definitions for an External Smart Meter" beschrieben ...?!

wolfganghuse, 2024-10-04:
Dieser Link gibt etwas Aufschluss. Der externe Messkopf wird bei Leitungen über 63A verwendet. 
https://support.huawei.com/enterprise/de/doc/EDOC1100338656/5c79c41a

cksoft0807, 2024-10-04:
Ich habe bei mir die Register einfach addiert. Das scheint bisher zu funktionieren. Beides wird nicht vorhanden sein.

bolliy, 2024-10-09:
So, die **erste Entwicklerversion der EMMA Integration** für den sun2000 Adapter ist fertig.

Da ich selbst keine EMMA habe, konnte ich die Umsetzung nur im Blindflug ohne Testing programmieren!
Zur Zeit wird nur der build-in meter sensor abgefragt. 
Sofern die Abfrage der EMMA prinzipiell funktioniert, können ggf. weitere Register eingebunden werden. 

_Voraussetzungen:_
    • **Installer acces**s/support für EMMA 
    • Die neuste EMMA firmware (**V100R024C00SPC100** oder höher) 
    • Verbinde EMMA mit dem lokalen LAN, WLAN scheint nicht zu funktionieren.
    • Verwendet die physische LAN-Verbindung des **WAN-Ports** an der EMMA unten, NICHT das LAN
    • ioBroker sollte im selben (V)LAN sein wie EMMA 
    • Aktivierung von Modbus auf der EMMA: Via the Fusion Solar app with the installer under Services > "Commissioning the device ” in the Settings > Communication configuration > **Enable Modbus TCP** connection, **TLS off** and setze die **ioBroker ip Adresse** als client IP address. 
Quelle: https://github.com/wlcrs/huawei_solar/issues/498#issuecomment-2336223081

_Installation:_
Die Installation erfolgt über den Expertenmodus. Danach auf die „Krakenkatze“ klicken und dann die benutzerdefinierte Url
https://github.com/bolliy/ioBroker.sun2000/tarball/emma
eingeben und die Installation starten.
Danach muss manuell der Adapter einmal neu gestartet werden! 

_Konfiguration:_
Reiter „Allgemeine Einstellungen“:
**Geräte-Ip-Adresse**: [ip Adresse der EMMA]
Sdongle abfrage: NEIN 
Unter dem Reiter „Integration“:
**Huawei Device:** Emma

und SPEICHERN…

Gruß Stephan

bolliy, 2024-10-09:
die neue EMMA Firmware V100R024C00SPC101 soll nun auch die modbus Anbindung per WLAN ermöglichen.
https://support.huawei.com/enterprise/en/fusionsolar/smarthems-pid-256539983/software/263209812

LG Stephan

gmihalca, 2024-10-09:
Hallo Stephan,

als erstens sorry das ich die EMMA noch nicht verschickt habe (hab die 2-te erst vor kurze Zeit von Lieferanten bekommen)

hab mal die Settings eingestellt und JA es kommen Daten.

Falls ich mit irgendwas helfen kann lass mich einfach wissen.

Lg
Gheorghe

https://support.huawei.com/enterprise/en/fusionsolar/smarthems-pid-256539983/software/263209812

wolfganghuse, 2024-10-09:
Vielen Dank für die Implementierung. Mein Installateur lässt mich aktuell mit der notwendigen Änderung hängen 🤷🏻‍♂️

bolliy, 2024-10-09:
@gmihalca das hört sich super an!
Für die Emma musste ich einen neuen Datentyp (readUnsignedInt64: "U64") basteln. Das Register 30302  "Inverter total absorbed energy" wird als Datentyp U64 gelesen und im Object "emma.inverterTotalAbsorbedEnergy" gespeichert. 
Würdest du/ihr bitte überprüfen ob der Wert plausible ist.
Läuft der Adapter mit der EMMA Integration stabil? 
Funktioniert die SmartLogger Integration noch einwandfrei? 
Sind alle Werte korrekt auch im Pfad "collected"?

LG Stephan

Rette70, 2024-10-09:
Hallo, Ich habe bei dem Adapter laufend diese Fehlermeldungen . Und IObroker wird dadurch instabil. Ansonsten kann ich den Akku S1 auslesen. Wechselrichter ist der M1 und eine EMM-02 verwaltet.

Error while reading from 192.168.178.132 [Reg: 47415, Len: 4, modbusID: 3] with: Modbus exception 3: Illegal data value (value cannot be written to this register)
--
sun2000.0 | 2024-10-09 20:38:30.177 | warn | Error while reading from 192.168.178.132 [Reg: 37100, Len: 38, modbusID: 3] with: Modbus exception 3: Illegal data value (value cannot be written to this register)
sun2000.0 | 2024-10-09 20:38:30.116 | warn | Error while reading from 192.168.178.132 [Reg: 37113, Len: 2, modbusID: 3] with: Modbus exception 3: Illegal data value (value cannot be written to this register)
sun2000.0 | 2024-10-09 20:38:21.975 | warn | Not all data can be read! 
Please inspect the sun2000 log.

Gibt es dazu Hilfe? Im Modbus protokoll von der EMMa finde ich diese Register auch nicht. Kann man diese abfragen löschen? Wenn ja wo. Vielen Dank. LG Mario

bolliy, 2024-10-09:
@Rette70 das sind Meter-Daten. Die sollten eigentlich bei der EMMA Integration nicht abgefragt werden.
Was hast du unter dem Reiter **Integration** eingestellt?

Rette70, 2024-10-09:
wo ist der? Ich hab nix eingestellt nur Port 502 und ID3

bolliy, 2024-10-09:
@Rette70 bitte das hier https://github.com/bolliy/ioBroker.sun2000/issues/63#issuecomment-2402612758 erste lesen.
Du musst die Entwicklerversion installieren ... dann klappt es auch mit dem Nachbarn ;)
Stephan

Rette70, 2024-10-09:
Danke, das war mir nicht bewusst, das es eine Erweiterung gibt. 
Jetzt bekomme ich aber noch diese Fehlermeldungen.

 [Reg: 30222, Len: 10, modbusID: 0] with: Modbus exception 3: Illegal data value (value cannot be written to this register)

 [Reg: 30000, Len: 43, modbusID: 0] with: Modbus exception 3: Illegal data value (value cannot be written to this register)

Danke für deine Hilfe

bolliy, 2024-10-09:
@Rette70 die habe ich zum Testen erst vor einigen Stunden freigegeben.
Welche EMMA Firmware hast du installierst?

Rette70, 2024-10-09:
die neuste 101

bolliy, 2024-10-09:
OK, dann muss ich die Abfrage nochmal anpassen ...
Installiere mal die Entwicklerversion neu ...

Rette70, 2024-10-09:
Bleibt bei 30000 und 30222 als Fehler

bolliy, 2024-10-09:
@Rette70 da ich keine Emma habe ist das etwas schierig die Fehler zu beheben.
Tauchen die Fehler einmal oder immer wieder auf?

Rette70, 2024-10-09:
glaube ich. Bin für jede Hilfe dankbar. 
dauerhaft alle 10 sec

bolliy, 2024-10-09:
Ok, werde mal morgen darüber nachdenken....
Feierabend.

LG Stephan

bolliy, 2024-10-10:
Hi @Rette70 habe nun eine neue Version für dich. Falls die Fehler weiterhin bestehen, sollte die Fehlermeldungen nur einmal im Log auftauchen!

Installation:
Die Installation erfolgt über den Expertenmodus. Danach auf die „Krakenkatze“ klicken und dann die benutzerdefinierte Url
https://github.com/bolliy/ioBroker.sun2000/tarball/emma
eingeben und die Installation starten.
Danach muss manuell der Adapter einmal neu gestartet werden!

Stephan

Rette70, 2024-10-10:
Hallo Stephan, also im Moment kommen keine Fehlermeldungen mehr. Danke. Gibt es eine Doku (Link) für die ganzen Datenpunkte. Das Huawei Datenblatt von der Emma habe ich.
Viele sind ja nicht selbsterklärend. Ansonsten muss ich mal durchschauen, wo die Anleitungen sind.
Mich würde der Datenpunkt momentaner Hausverbrauch interessieren. Den messe ich zwar im Moment mit Shelly 3EM aber ein Vergleich zu EMMA wäre interessant. 
Vielen Dank LG Mario

wolfganghuse, 2024-10-10:
Die sind doch auf Seite 10 des Modbus Dokuments aufgeführt… Voltage und Current für die einzelnen Phasen. 
Ich hab jetzt für den 16. einen Termin um endlich die Emma konfiguriert zu bekommen.

Rette70, 2024-10-10:
> Die sind doch auf Seite 10 des Modbus-Dokuments aufgeführt… Spannung und Strom für die einzelnen Phasen. Ich hab jetzt für den 16. einen Termin um endlich die Emma zu konfigurieren bekommen.

Ja, hab das nur alles in englisch. Und bei den Übersetzungen kommt oftmals was anderes raus.

Slash-12-de, 2024-10-10:
@bolliy: Vielen Dank schonmal für die ganze Arbeit und die super Implementierung. Sieht schon sehr gut aus. 
Bei mir läuft die neueste Version und ich erhalte auch noch einen Fehler, welche die Funktion aber scheinbar nicht beeinträchtigen.
Es kommt im 20 Sekundentakt (was genau dem Aktualisierungsintervall entspricht) zuerst der folgende Fehler und danach die darunter stehende Warnung.

sun2000.0 | 2024-10-10 21:00:40.010 | error | Can not connect to Huawei inverter for modbus ID 1!

sun2000.0 | 2024-10-10 21:00:40.010 | warn | Error while reading from 192.168.10.41 [Reg: 30000, Len: 71, modbusID: 1] with: Modbus exception 4: Slave device failure (device reports internal error)

Muss ich hier die Settings noch anpassen? Ich habe nur die Emma mit der xxx100 Firmware im Einsatz und kein SmartLogger oder ähnliches. Vielen Dank

bolliy, 2024-10-11:
@Slash-12-de der Adapter erkennt nicht eigenständig welche Hardware installiert ist. 
Bitte unter dem Reiter Integration die EMMA auswählen. 

_Konfiguration:_
Reiter „Allgemeine Einstellungen“:
Geräte-Ip-Adresse: [ip Adresse der EMMA]
Sdongle abfrage: NEIN
Unter dem Reiter „**Integration**“:
Huawei Device: **Emma**

Stephan

Slash-12-de, 2024-10-11:
> @Slash-12-de der Adapter erkennt nicht eigenständig welche Hardware installiert ist. Bitte unter dem Reiter Integration die EMMA auswählen.
>
> _Konfiguration:_ Reiter „Allgemeine Einstellungen“: [...]

@bolliy: Die Einstellungen hatte ich bereits, wie von dir beschrieben hinterlegt, daran scheint es nicht zu liegen. In den Objekten wird auch die Emma, mit Firmware, ... korrekt aufgeführt.

bolliy, 2024-10-11:
@Slash-12-de die Fehlermeldung "Can not connect to Huawei inverter for modbus ID 1" sagt aus, dass der WR nicht auf der modbus ID 1 über EMMA erreichbar ist. 
Ist die modbus Kommunikations-ID des Wechselrichters wirklich 1 ? 
Siehe hier: https://github.com/bolliy/ioBroker.sun2000/blob/main/docs/inverter.md

Die Vorzeichen von ActivePower schau ich mir gleich an…

LG Stephan

bolliy, 2024-10-11:
So @Slash-12-de habe nun eine neue Entwicklerversion hochgeladen.
Nun wird ActivePower mit -1 multipziert. In der Modbus Definition steht leider nichts.  Leider wäre dieses Verhalten nicht konsistent zum sun2000 und SmartLogger von Huawei.

Installation:
Die Installation erfolgt über den Expertenmodus. Danach auf die „Krakenkatze“ klicken und dann die benutzerdefinierte Url
https://github.com/bolliy/ioBroker.sun2000/tarball/emma
eingeben und die Installation starten.
Danach muss manuell der Adapter einmal neu gestartet werden!

LG Stephan

bolliy, 2024-10-11:
@Rette im Wiki und im Readme sind einige Datenpunkte beschrieben:
https://github.com/bolliy/ioBroker.sun2000/wiki
https://github.com/bolliy/ioBroker.sun2000/blob/main/docs/README.md

LG Stephan

Slash-12-de, 2024-10-11:
@bolliy : Vielen Dank für die prompte Anpassung. Jetzt sieht es sehr gut aus.

bolliy, 2024-10-11:
@Slash-12-de was war mit der modbus ID?

Slash-12-de, 2024-10-11:
> @Slash-12-de was war mit der modbus ID?

Dazu hatte ich weiter oben gepostet, dass das Problem war, dass ich die Wechselrichter ID 2 hinterlegen musste, damit war der Fehler behoben. Also so wie von dir beschrieben 😀. Danke

bolliy, 2024-10-11:
Ich verlege nun die Diskussion ins Forum https://forum.iobroker.net/topic/71768/test-adapter-sun2000-v0-1-x-huawei-wechselrichter
Vielen Dank für eure Unterstützung!

LG Stephan

bolliy, 2024-10-20:
@gmihalca konntest du die Emma mit dem Adapter schon testen?
Sofern die Integration funktioniert, bitte das Issue schließen.

LG Stephan

closed 2024-10-20 by gmihalca

cksoft0807, 2024-10-28:
Ich würde gerne noch einmal beim Thema "meter/activePower" einhaken. Wäre nicht eigentlich der Wert Sampled Data, "Feed-in power" (30358) das passende Register für die Anlage? Müsste dann auch das Problem interner/externer Sensor lösen.

Die Hauslast sollte man mit Emma eigentlich auch ohne Berechnung über "Load power, 30356" abgreifen können?

[1 more reply]