bolliy/ioBroker.sun2000 #72: [Bug]: Adapter wacht nach wenig Sonneneinstrahlung nicht mehr auf

arnovy, 2024-04-03:
### The problem

Hi Stephan,
der Adapter 0.61 ist heute um 8:06 wegen zu wenig Sonneneinstrahlung wieder schlafen gegeangen, allerdings nie mehr aufgewacht.

Nach drei Stunden habe ich ihn dann neu gestartet, danach lief alles wieder.

wäre es möglich den irraditionmode einfach zu deaktivieren? Längen und Breitengrad im Broker sind gesetzt. 

### Version of Adapter

NA

### Repository

ioBroker beta

### Checklist of files to include below

- [ ] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

bolliy, 2024-04-03:
Ich habe noch eine Änderung vorgenommen. 
Bitte auf v0.6.2 updaten.
Siehe https://forum.iobroker.net/post/1145740

Stephan

bolliy, 2024-04-17:
Sofern das Problem gelöst wurde, würde ich diesen issue gerne schließen. 
Ist das in Ordnung?

arnovy, 2024-04-17:
Hi Stephan,

das Problem trat mit 0.6.2 nicht mehr auf, kann geschlossen werden

closed 2024-04-17 by bolliy