bolliy/ioBroker.sun2000 #73: [Bug]: Modellname wird nicht korrekt ausgelesen

RoHa78, 2024-04-07:
### The problem

Der Modellname wird nicht korrekt ausgelesen, bzw. nicht richtig umgesetzt.
Ich habe das Modell SUN2000-8KTL-M1, durch einen mir nicht nachvollziehbaren Grund wird der Originalmodellname mit "SUN2000-10KTL-" ergänzt, sodass am Ende die Zeichenfolge "SUN2000-8KTL-M1SUN2000-10KTL-" als Wert hinterlegt ist
[image]

### Version of nodejs

v18.17.1

### Version of ioBroker js-controller

5.0.19

### Version of Adapter

0.7.0

### Repository

ioBroker beta

### Operating system running ioBroker

Debian version: 11 (bullseye) (Raspberry)

### Checklist of files to include below

- [ ] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

bolliy, 2024-04-08:
Ich habe den Fehler lokalisiert. Wird beim nächsten update gefixt.
LG Stephan

RoHa78, 2024-04-08:
Super, vielen Dank!
Gruß  Ronald

bolliy, 2024-04-09:
Sollte nun funktionieren.
Hab gerade die v0.7.1 auf rpm veröffentlicht. Kann morgen auch über das beta repository installiert werden. 
Falls der Fehler behoben wurde, bitte diesen issue schließen.
Lg Stephan

RoHa78, 2024-04-09:
Perfekt, nun passt es.
Viele Grüße Ronald

closed 2024-04-09 by RoHa78