bolliy/ioBroker.sun2000 #75: [Bug]: Unrealistische Werte während SW Update (Inverter)

MM1834, 2024-04-10:
### The problem

Auch wenn es unter Umständen keinen Sinn macht den Adapter während SW Upgrades aktiv zu belassen so registriert der sun2000 Adapter unbrauchbare Werte. Unter Umständen gibt es ein Flag welches Upgrade/Maintenance signalisiert damit der Adapter keine Werte ausgibt oder einen validation-check ausführt.

1) LUNA-Upgrade (V100R002C00SPC125 -> V100R002C00SPC127): Im aktuellen Fall trat der Effekt mit den ungültigen Werten beim Upgrade der Luna2000-Software auf daher liegt die Vermutung nahe, dass es es eher mit modbus Auslastung zusammen hängt. Nach dem Upgrade zeigt der Adapter ohne Neustart die korrekten Werte an.

2) Dongle-Upgrade: Gemäss Huawei Release Notes ist aktuell auch eine neue Version für den Dongle bzgl. modbus issues verfügbar - die war in vorliegenden Fall noch nicht installiert (IST: V200R022C10SPC108, Geplant: V200R022C10SPC114). Der Upgrade der Dongle-SW produzierte adapter timeout, was auch verständlich ist, aber keine ungültigen Werte.

3) Inverter-Upgrade (V100R001C00SPC159 -> V100R001C00SPC162; nach Dongle-Upgrade): Nur einige 0-Werte (SOC Luna2000) nach Abschluss des Inverter-Upgrades. Danach zurück auf den korrekten Wert ohne Neustart des Adpaters.
"2024-04-10 18:48:45.375	warn	Error while reading from <IP> [Reg: 32080, Len: 2, modbusID: 1] with: Modbus exception 4: Slave device failure (device reports internal error)"

### Version of Adapter

0.71

### Repository

ioBroker beta

### Checklist of files to include below

- [ ] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

[image]
[image]
[image]
[image]

bolliy, 2024-04-10:
Man kann den Upgrade/Maintenance Zustand ggf. über den device Status Reg 32089 ablesen. In der Interface Definition habe ich sonst kein passendes Register gefunden.
Welchen Wert hatte der State `sun2000.0.inverter.0.deviceStatus` ?  Dann kann ich darauf reagieren.

LG  Stephan

bolliy, 2024-04-11:
Ich werde bei der nächten Beta-Version versuchen die Werte einzufangen. Kann aber dauern, da ich erst die v0.7.1 ins stable Repository bringen will.
Stephan

Obsthaendler, 2024-04-12:
Hallo,

ich habe eben auch mein SDongle von V200R022C10SPC108 auf V200R022C10SPC114 aktualisiert und dabei die Werte in iobroker beobachtet. Bei mir gab es kein Problem der Werte. 

Bezüglich eine if-Abfrage für den Status Upgrade konnte ich auch nichts finden.

Anbei das Log während und nach dem Upgrade des SDongles:

`
sun2000.0
2024-04-12 08:12:24.725	info	Connected Modbus TCP to 192.168.2.16:502

sun2000.0
2024-04-12 08:12:19.722	info	Open Connection...

sun2000.0
2024-04-12 08:12:19.722	warn	Error while reading from 192.168.2.16 [Reg: 32064, Len: 2, modbusID: 1] with: Modbus exception 4: Slave device failure (device reports internal error)

sun2000.0
2024-04-12 08:11:15.273	info	The inverter with modbus ID 2 is accessible again.

sun2000.0
2024-04-12 08:11:15.273	info	The inverter with modbus ID 1 is accessible again.

sun2000.0
2024-04-12 08:09:58.393	info	The inverter with modbus ID 2 is no longer accessible. That is why the logs are minimized.

sun2000.0
2024-04-12 08:09:58.393	info	The inverter with modbus ID 1 is no longer accessible. That is why the logs are minimized.

sun2000.0
2024-04-12 08:09:58.392	warn	Error while reading from 192.168.2.16 [Reg: 37113, Len: 2, modbusID: 1] with: connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:58.390	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:58.387	info	Open Connection...

sun2000.0
2024-04-12 08:09:56.386	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:56.383	info	Open Connection...

sun2000.0
2024-04-12 08:09:56.383	warn	Error while reading from 192.168.2.16 [Reg: 32064, Len: 2, modbusID: 1] with: connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:56.381	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:56.379	info	Open Connection...

sun2000.0
2024-04-12 08:09:54.377	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:54.371	info	Open Connection...

sun2000.0
2024-04-12 08:09:54.370	warn	Error while reading from 192.168.2.16 [Reg: 32080, Len: 2, modbusID: 1] with: connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:54.368	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:54.363	info	Open Connection...

sun2000.0
2024-04-12 08:09:52.360	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:52.356	info	Open Connection...

sun2000.0
2024-04-12 08:09:52.353	warn	Error while reading from 192.168.2.16 [Reg: 32064, Len: 2, modbusID: 2] with: connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:52.351	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:52.349	info	Open Connection...

sun2000.0
2024-04-12 08:09:50.345	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:50.333	info	Open Connection...

sun2000.0
2024-04-12 08:09:50.332	warn	Error while reading from 192.168.2.16 [Reg: 32080, Len: 2, modbusID: 2] with: connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:50.325	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 connect ECONNREFUSED 192.168.2.16:502

sun2000.0
2024-04-12 08:09:50.322	info	Open Connection...

sun2000.0
2024-04-12 08:08:50.006	warn	Couldnt connect Modbus TCP to 192.168.2.16:502 TcpRTUBufferedPort Connection Timed Out

sun2000.0
2024-04-12 08:08:40.003	info	Open Connection...

sun2000.0
2024-04-12 08:08:40.002	warn	Error while reading from 192.168.2.16 [Reg: 32080, Len: 2, modbusID: 1] with: Timed out`

VG
Steffen

bolliy, 2024-04-12:
Hallo Steffen,
danke für deinen Test. 
Ich werde einfach eine Funktion implementieren, die abhängig vom Datentyp den Wert auf Plauibilität prüft. Falls der Wert nicht im erlaubten Wertebereich liegt, wird dieser auf 0 gesetzt. Das sollte das Problem lösen.
Ich sehe abger mit Freude, dass der Adapter nun ganz gut mit den Verbindungsproblemen umgeht. Er schaltet sich einfach auf standby - egal ob der Inverter selbst in standby geht oder ein Verbindungsaufbau nicht mehr zustande kommt.
 
LG Stephan

bolliy, 2024-04-22:
Seit der v0.8.0 sollten das Problem gelöst sein.
Sofern es beim nächsten Firmware update keine unplausiblen Werte auftauchen, bitte ich diesen issue zu löschen.

LG Stephan

MM1834, 2024-04-23:
Super und Danke Stephan. Werde die Werte beim nächsten Huawei Upgrade, insbesondere einem für die LUNA weiter beobachten und je nach Situation den Issue löschen oder kommentieren.

MM1834, 2024-06-12:
Beim heutigen Upgrade des Battery- und des MODBUS-Modules hat der Adapter ungültige Werte (ausserhalb des definierten Bereichs) im Log wie erwartet als Warnung gemeldet und ignoriert da sie.
Ich denke der Issue kann geschlossen werden.

bolliy, 2024-06-13:
@MM1834 Vielen Dank für deine Unterstützung.

closed 2024-06-13 by bolliy