bolliy/ioBroker.sun2000 #79: Refresh Meter even when in Standby

Pendragon77, 2024-05-04:
### What would you like to be added?

In the documentation is stated: 

> Sofern an dem in standby befindlichen sun2000 WR auch noch der Smart Power Sensor ageschlossen ist, aktualisieren sich die Werte im Pfad sun2000.0.meter ebenfalls nicht.

 I would like to have these values updated with the high frequency, even when the inverter is in standby.

### Why is this needed?

I am using the meter as a source for controlling my Wallbox and home automation. Til now I was using [solaranzeige.de](solaranzeige.de) for the visualisation, which even reads the meter at night.

Now I tried your proxy today and forwarding the values to my homeautomation, but the values stopped at sundown and and no further processing could be made.

Please consider a change.

bolliy, 2024-05-05:
Ich gehe davon aus, dass du keinen Speicher hast. Ohne Speicher geht der WR in den Standby mode: **detecting irradiation**. Nach meinem Kenntnisstand ist die modbus-Kommunikation bei diesem standby mode komplett runter gefahren. Deshalb unterbleiben jegliche Anfragen an den WR seitens des Adaptes. Ansonsten hagelt es nur TimeOuts.
Da mein WR nicht in den Standby: detecting irradiation geht (ich habe eine Luna 10kWh) , kann ich das natürlich nicht nachvollziehen. 
Wenn du allerdings andere Informationen hast, kann ich das gerne im Adapter anpassen. 
Dann bräuchte ich allerdings deine Hilfe - du müsstest einige Tests durchführen.

LG Stephan

Pendragon77, 2024-05-05:
Hi Stephan,
kein Problem mit dem Testen. Ich bekomme das alles gut parallel installiert. Ich habe den SUN2000-12KTL-M2 ohne Akku, aber mit Dongle.
Es würde ja wohl reichen, wenn der WR im DeviceStatus<4 ist, nur den Bereich ab 37100 auszulesen.
Aktuell habe ich solaranzeige laufen. Das Skript läuft auf php und zieht auch nachts die Daten, weshalb ich gestern abend dann das wieder als Quelle angeschaltet habe.

Eben gerade habe ich diesen Code durchgesehen und nichts gefunden, was auf eine Standby Detektion hindeuten würde. Vielleicht verhalten sich die verschiedenen WR Generationen anders?
Wenn ich es schaffe, versuche ich mal bei Dunkelheit die Mosbus Register manuell auszulesen und zu schauen was kommt.

Besten Dank
Christian

bolliy, 2024-05-05:
Hi Christian,
ich glaube an deinem M2 kann man keinen Speicher anschließen.  
Insofern kann das durchaus sein, dass sich der M2 im standby anders verhält …
Würdest du bitte neben den **meter daten** auch **Activ Power** und **Input Power** abfragen.:

```
{
				address : 32080,
				length : 2,
				info : 'Inverter Activ Power',
				refresh : dataRefreshRate.high,
				type : deviceType.inverter,
				states : [{
					state: {id: 'activePower', name: 'Active power', type: 'number', unit: 'kW', role: 'value.power.active', desc: 'reg:32080, len:2, Power currently used',},
					register: {reg: 32080, type: dataType.int32, gain:1000},
					store : storeType.always
				}]
},
{
				address : 32064,
				length : 2,
				info : 'Input Power',
				refresh : dataRefreshRate.high,
				type : deviceType.inverter,
				states : [{
					state: {id: 'inputPower', name: 'Input power' , type: 'number', unit: 'kW', role: 'value.power.produced', desc: 'reg:32064, len:2, Power from solar'},
					register: {reg: 32064, type: dataType.int32, gain:1000},
					store : storeType.always
                                 }]
}
```

Sofern also der M2 im standby auch Daten über den modbus ausliefert, könnte ich ein eigenen Treiber diesen Typ bauen. 
Ich bin gespannt auf dein Ergebnis.
LG Stephan

bolliy, 2024-05-05:
Übrigens: Ich hatte auch mal solaranzeige.de bei mir laufen. Ich fand allerdings die Umsetzung in php für meine Zwecke nicht optimal. In node.js ist es deutlich einfacher Echtzeitanwendungen zu realisieren. Ich habe mir nun alles in ioBroker mit node.js gebastelt. Für mich ist ioBroker die vielseitigste IOT-Plattform!

Pendragon77, 2024-05-05:
> Übrigens: Ich hatte auch mal solaranzeige.de bei mir laufen. Ich fand allerdings die Umsetzung in php für meine Zwecke nicht optimal. In node.js ist es deutlich einfacher Echtzeitanwendungen zu [...]

Ja, das sehe ich auch. Ulrich, der Admin von Solaranzeige konnte mir anfang des Jahres die Huawei Integration noch schnell aufbohren, weil die Meter-Daten nicht drin waren. 
Solaranzeige war, bis ich dein Script gefunden habe,  die einfachste Möglichkeit die Daten auszulesen. Die kommen dann in meine separate Inlfux-DB und werden per MQTT in Edomi weiterverartbeitet. Da ich aus der KNX-Welt komme, war das die Wahl der Automation. IOBroker empfehle ich jetzt für Anfänger als Plattform.

Pendragon77, 2024-05-05:
Es läuft! Links immer Tags und rechts während der Standby-Phase:
[image]
[image]

In 32089 ist der Status, der sich von Grind auf "Detecting Radioation" ändert. Die Werte vom Meter scheinen auch alle befüllt zu werden. _Am Tag habe ich nicht lange laufen lassen, um zu sehen ob alles kommt_

Beste Grüße
Christian

bolliy, 2024-05-05:
Christian, wenn ich das richtig sehe, dann funktioniert das modbus device des M2 inverters auch im standby mode. 
Ich werde dir in den nächsten Tagen eine Entwickler-Version bauen. Diese wird bei device-status = 2 nicht mehr die Abfrage an den WR unterbinden.

Ich melde mich wieder, sofern ich das umgesetzt habe.

LG Stephan und einen schönen Abend wünsche ich.

Pendragon77, 2024-05-05:
Perfekt! Vielen Dank!

Evtl. macht es Sinn die Model ID aus Register 30000 als Abfrage-Bedingung zu nehem, welche Wechselrichter Nachts arbeiten. Dann braucht es nicht verschiedene Version des Adapters. Danach muss nur noch herausgefunden werden, ob es eine Systematik gibt, nach der man das unterscheiden kann.
Dir auch nen schicken abend noch!
Christian

bolliy, 2024-05-05:
Es wird keine unterschiedlichen Versionen des Adapters geben. Ich verfolge einen objektorientierten Ansatz. Es wird eine weitere inverter Klasse M2 geben, die die Eigenschaften und Methoden von dem M1/M0 erbt. Allerdings werden ggf. einige Methode überschrieben.
Stephan

Pendragon77, 2024-05-06:
Hi Stephan,
das ging schnell.
Daten auslesen geht super!
Der Adapter startet ab und zu recht langsam (das Icon bei "Verbunden mit Gerät oder Dienst" bleibt lange gelb). Als ich dann im Debug Modus nachgeschaut habe, kamen aber Daten. Ein paar Timeouts konnte ich auch beobachten, aber das könnte ich dann mal genauer beobachten.

Was jetzt nicht mehr funktioniert, ist der Modbus-Proxy. Solaranzeige bekommt die Daten vom WR, wenn ich aber Umschalte (Deinen Adapter starten und WR_IP von Solaranzeige auf IOBRoker ändere, dann kommen keine Daten mehr an. Erweiteres logging hat in den protokollen nichts zusätzliches angezeigt.
Es muss nicht 100% an dem Adapter liegen, da ich es gestern abend auch nicht lassen konnte an den VLAN Einstellungen des Dockers rumzuspielen. Nun hat IOBroker eine eigen IP und sollte eigentlich weniger stress machen als vorher...

Vielleicht fällt dir dazu ja direkt was auf. Ansonsten kann ich gerne Daten und logs nach Bedarf liefern.

Beste Grüße
Christian

bolliy, 2024-05-06:
Hallo Christian,

erst nach 1 Minute und wenn alle Daten einmal gelesen wurde schaltet der Adapter auf grün.
Bei mir funktioniert der modbus-proxy noch. Ich habe eine weitere adapter instanze, die die daten über den proxy bekommt.
Zu den timeouts warten bis licht wieder auf deine molule fällt - vielleicht sind dann die timeouts dann weg. 

timeouts können mehrere ursachen haben.

LG Stephan

Pendragon77, 2024-05-06:
Japp, hast recht.
Habe den alten Docker von gestern Nachmittag nochmal getestet und von dort kommen Daten mit den Netzwerkeinstellungen...
Dann muss ich nochmal bei mir suchen. Danke nochmal für die Anpassung!!

bolliy, 2024-05-08:
Hi Christian,
wie sieht es aus - funktioniert alles?
LG Stephan

Pendragon77, 2024-05-08:
Hi Stephan,
der Adapter läuft jetzt durch und keine Aussetzer. Meiner Ansicht nach, ist alles stabil!

bolliy, 2024-05-08:
Dann wird es bald ein neues Release (beta) geben.
Christian, vielen Dank für deine Unterstützung!

VG Stephan

bolliy, 2024-05-15:
In der Beta-Version v0.9.0 wurde die Abfrage der Registerdaten für die Serie M2, M3 im standby mode neu geregelt.

closed 2024-05-15 by bolliy