bolliy/ioBroker.sun2000 #84: Please add 'Backup power SOC' as control parameter

max19751, 2024-06-10:
### What would you like to be added?

Please add 'Backup power SOC' as parameter to read and write

### Why is this needed?

In systems, where a battery and a backup box are installed, the 'Backup power SOC' defines down to what battery minimum level the consumption is limited in case of normal usage. The 'Backup power SOC' is hence the reserve left in the battery in case the system switches to backup power because of a power outage. 
Being able to set this parameter allows the user to define for their own usage scenario how much of the battery is reserve. As this might depend on weather forecast etc, it is a parameter that will be set more frequently when optimising the battery usage.

bolliy, 2024-06-10:
Hi Max,
Do you mean the register address 47102?
[image]

max19751, 2024-06-10:
Hi, yes, indeed, this is what I mean.

bolliy, 2024-06-14:
Hi Max, dein issue ist umgesetzt. 
LG Stephan

max19751, 2024-06-23:
Hi Stephan, vielen Dank! Gestestet, funktioniert.
LG Max

closed 2024-06-23 by max19751