bolliy/ioBroker.sun2000 #85: Batterypack-Daten ergänzen

besimo1, 2024-06-12:
### What would you like to be added?

**Bitte folgende Batterie-Register ergänzen (evtl. als Option auswählbar in den Instanzeinstellungen):**
**37814** [Energystorage unit1] **Softwareversion**
37920, 37921, 37922, 37921 [Energystorage unit1][Batterypack 1,2 und 3] SOH Calibration Status
**38210, 38252, 38294, 38336, 38378, 38420** [Energystorage unit1,2][Batterypack 1,2 und 3] **Firmwareversion**
38228, 38270, 38312, 38354, 38396, 38438 [Energystorage unit1,2][Batterypack 1, 2 und 3] Workingstatus
**38229, 38271, 38313, 38355, 38397, 38439**  [Energystorage unit1,2][Batterypack1, 2 und 3] **SOC**
**38223, 38275, 38317, 38359, 38401, 38443**   [Energystorage unit1,2][Batterypack 1, 2 und 3] **Charge/Dischargepower**
**38235, 38277, 38319, 38361, 38403, 38445**  [Energystorage unit1,2][Batterypack 1, 2 und 3] **Voltage**
38236, 38278, 38320, 38362, 38404, 38446   [Energystorage unit1,2][Batterypack 1, 2 und 3] Current
**38238, 38280, 38322, 38364, 38406, 38448**  [Energystorage unit1,2][Batterypack 1, 2 und 3] **Totalcharge**
**38240, 38282, 38324, 38366, 38408, 38450**  [Energystorage unit1,2][Batterypack 1, 2 und 3] **Totaldischarge**
38452, 38454, 38456, 38458, 38460, 38462  [Energystorage unit1,2][Batterypack 1, 2 und 3] Maximum temperature
38453, 38455, 38457, 38459, 38461, 38463  [Energystorage unit1,2][Batterypack 1, 2 und 3] Minimum temperature

(Hinweis Bei Unit2 ProductMode ist steht in der Erklärung reg. 37089, in den Modbus Interface
Definitions vom 16.02.2023 steht "47089" )

### Why is this needed?

**### Allg. Batterieüberwachung**
z.B. zur Erkennung, ob ein Garantiefall vorliegt (z.B. Kapazitätsverlust eines einzelne Batteriepacks - berechnet aus tatsächlicher Lade-/Entladekapazität des einzelnen Batteriepacks, da SOC stark schwankt).
Überwachung der Akkuspannung der einzelnen Batteriepacks - Profis erkennen an den Spannungen viel genauer, als am SOC, ob ein Problem vorliegt.
Überwachung der Firmewareversionen (mir wurde bereits zweimal ungefragt einmal eine neue und einmal eine ältere Firmware aufgespielt)=>historische_Daten der installierten Firmwarestände (Datum der Updates) möglich
Berechnung des Batteriewirkungsgrades (hier gab es bei v127 bei mir einen extremen Effizienzverlust gegenüber v119 und v125)
Überwachung der Temperatur (z.B. Begrenzung Lade- / Entladeleistung, wenn Temperatur eines Batteriepacks einen Schwellwert überschreitet zur Schonung der Batterie).
Mein Ziel ist es, FusionSolar irgendwann nicht mehr zu benötigen. Dazu fehlen mir noch die Daten der einzelnen Akkus.

bolliy, 2024-06-13:
@besimo1 ich kann sicherlich einige Batteriedaten zusätzlich aufnehmen. Nur sehe ich die Stabilität des Adapters bei der großen Anzahl von zusätzlichen Daten gefährdet. Wenn wir zu viele Daten in einem Zeitraum gelesen werden, dann wird der sDongle instabil. Gerade die Überwachung der Batteriepacks sehe ich kritisch. Zumal ich dann auch eine weitere Abfrageschleife bräuchte.
Ich möchte dir gerne bei deinem Garantiefall helfen. Könntest du die Anzahl der Register auf ein Minimum reduzieren?

besimo1, 2024-06-13:
Danke schon Mal im Voraus für die Hilfe.
Ich selbst habe den Luna-10-S1 mit zwei Batterypacks. Erläuft bei mir unter Unit 2. Deshalb brauche ich die Werte für Unit2 und Batterypack 1 und 2
Für die Überprüfung der Garantie wären am Wichtigsten in absteigender Reihenfolge:
- **38366, 38408 [Energystorage unit2][Batterypack 1, 2 ] Totaldischarge**
- **38364, 38406 [Energystorage unit2][Batterypack 1, 2 ] Totalcharge**
- **38361, 38403 [Energystorage unit2][Batterypack 1, 2 ] Voltage**
- 38355, 38397 [Energystorage unit2][Batterypack1, 2] SOC
- 38359, 38401 [Energystorage unit2][Batterypack 1, 2] Charge/Dischargepower
VG
Besimo

bolliy, 2024-06-15:
Hallo @besimo1 ich könnte dir eine eingebe Version basteln, die für dich die o.g. Register abfragt und als State speichert.
Würdest du im Gegenzug deine Auswertungen, Erfahrungen und den Verlauf deines Garantiefalls im Forum posten?
Ich glaube so könnten auch andere von deinen Erfahrungen profitieren.
Wie ist deine Haltung dazu?

Stephan

besimo1, 2024-06-16:
Hallo Stephan,
ich will nicht zu viel versprechen und weiß noch nicht, wieviele Details ich öffentlich von dem Garantiefall im Einzelnen posten will.
Ich habe so ein paar Ideen zur Programmierung, z.B. will ich mir ein Script erstellen, mit dem ich per Button die Restkapazität jedes einzelnen Speichermoduls durch erzwungenes Laden mit vorgegebener Lade-/Entladeleistung gemäß den Garantiebestimmungen bestimme (laut Huawei-Support (Stand vor einigen Monaten) gibt es in FusionSolar kein Programm für die Bestimmung der Restkapazität).
Wenn ich tatsächlich so ein Script erstellt habe, könnte ich es im Forum veröffentlichen und ein paar Worte / Erklärungen dazu schreiben und z.B. die Auswertung eines Kapazitätstests posten.
VG
Besimo

besimo1, 2024-06-16:
...
und es wäre schon gut, wenn alle 5 (bzw. 10) Werte / Register abgefragt werden würden, also auch SOC und die Lade-/Entladeleistung jedes Batterypacks, so dass man auch nachweisen kann, dass die Vorgaben der Garantiebedingungen auch beim Test für jeden einzelnen Akkupack eingehalten wurden.

bolliy, 2024-06-16:
ok, dann werde ich das in den nächsten Tagen angehen...

bolliy, 2024-06-20:
Hallo Besimo,
ich habe nun eine Version für dich gebaut.
Die Installation erfolgt über den Expertenmodus. Danach auf die „Krakenkatze“ klicken und dann die benutzerdefinierte Url
https://github.com/bolliy/ioBroker.sun2000/tarball/besimo1
eingeben und die Installation starten. Bitte nach der Installation den Adapter manuell neu starten!

Stephan

besimo1, 2024-06-23:
Vielen Dank Stefan für die exklusive Version für mich,

läuft bisher super und stabil, außer TotalDischarge wird nur 0 angezeigt:
[image]
(chargeDischargePower ist gerade null, da Akkkus voll sind)

Die Nummer 38666 und 38408 müssten aber passen laut Liste Modbus Interface Definition (Issue 5 vom 16.02.2023).

Ich habe mir ein Script geschrieben und mit Hilfe der Werte für chargeDischargePower für Akku1 und 2 die Werte im Minutentakt addiert. Da gibt es Abweichungen zu TotalCharge und Lade- und Entladekapazitaet des Speichers (currentDayChargCapacity und currentDayDischargeCapacity) und Fusion Solar.
Ich vermute, dass bei Batterie-Firmware v127 und v621 ein Fehler bei der Ermittlung der Entladeleistung vorliegt. Muss ich aber noch genauer beobachten.

zur Info meine Instanzeinstellungen:
[image] 
[image]
[image]
[image]

VG
Besimo

bolliy, 2024-06-23:
Hi @besimo1, ich glaube den Fehler bei mir gefunden zu haben. (date reg length was too short)
Installiere mal erneut und den Neustart nicht vergessen!

LG Stephan

besimo1, 2024-06-24:
Funktioniert !

bolliy, 2024-06-26:
Kann ich dein issue schießen?

besimo1, 2024-06-30:
Sorry für die späte Antwort (ich habe ein Zeit-Problem).

Sind dann die BatteryPack-Adressen auch in den zukünftigen Versionen enthalten - z.B. in v0.11 ?

Falls nicht, darf ich dann nicht mehr updaten (ich will diese Daten auch in Zukunft in meiner Visualisierung haben) ?

Mit den BatteryPack-Daten kann man die Batterieeffizienz bestimmen. Das ist wichtig, da es offensichtlich zwischen Versionen v119/v125 und v127/v621 deutliche Unterschiede gibt. Ich werde in den nächsten Tagen die Firmware wieder downgraden auf v125, da v621 an manchen Tagen deutlich ineffizienter ist, als v119/v125.

Hier ein Auszug aus meiner View:
[image]
[image]

bolliy, 2024-07-01:
Zur Zeit sind die Änderungen in der v0.11.0 zu v0.10.0 minimal. Also für dich nicht von Relevanz.
Ich kann aber in der nächsten minor release einen versteckten Schalter einbauen. Dann kannst du deine BatteryPack Daten einschalten. 
Sofern es soweit ist, werde ich dich hier informieren.
Dein issue bleibt bis dahin weiter offen....

besimo1, 2024-07-02:
Danke, dass du extra für mich einen versteckten Schalter einbauen willst.

Grundsätzlich würde ich es natürlich begrüßen, wenn der Schalter für alle zugänglich wäre und alle Akkukonfigurationen (unit 1 + 2, Akku 1 bis 3) möglich wären.

Ich weiß nicht, inwieweit sich andere User für dieses Feature interessieren. Vielleicht schreiben hier andere User, ob sie an diesem Feature interessiert sind, solange das issue offen ist.

Für mich allein reicht natürlich der versteckte Schalter.

fuStefan, 2024-07-16:
Hi, für mich sind die zusätzlichen Daten auch sehr interessant, einfach nur um sie zu loggen und die Entwicklung über die Jahre zu beobachten. Vielleicht könntest du einen offiziellen Schalter in der Adapterkonfiguration vorsehen.

bolliy, 2024-09-23:
In der v0.12 sind nun die Battery Packs enthalten.

https://forum.iobroker.net/topic/71768/test-adapter-sun2000-v0-1-x-huawei-wechselrichter/547?_=1727103674960

LG Stephan

bolliy, 2024-09-25:
@besimo1 ich würde gerne das issue schließen. Ist das ok für dich?

closed 2024-09-29 by bolliy