bolliy/ioBroker.sun2000 #86: Support for new Battery System Luna2000-21-S1

fritzrichter, 2024-06-13:
### What would you like to be added?

I installed the new Luna-S1 Battery System, which is not identified right now, I guess new registers are used. We could work together if you reach out and change the plugin so that the metrics are imported correctly.

### Why is this needed?

Would like to see all KPIs as well in the system.

bolliy, 2024-06-13:
Do you have the modbus definitions for your new storage? Otherwise I can't implement your issue.

bolliy, 2024-06-13:
Is the **battery** path behind the inverter empty or has data been written there?

bolliy, 2024-06-13:
If you don't mind, we can continue in German...

fritzrichter, 2024-06-13:
Alles klar ;) ich sammle mal mehr Infos und versuche die Register manuell zu lesen und gebe Bescheid… Danke für die schnelle Antwort ;)

fritzrichter, 2024-06-14:
Ich habe den Fehler auf meiner Seite gefunden! Nach einer Anpassung verhält sich der Luna S1 nun so wie der alte und die Werte werden korrekt importiert. Großartig :) Schließe das Ticket daher.

closed 2024-06-14 by fritzrichter