bolliy/ioBroker.sun2000 #98: [Bug]: probleme beim Update

luckyheiko, 2024-07-08:
### The problem

Hallo 

weiß nicht ob es an mir (DAU) liegt oder was anderem, aber ich kann gerade nicht auf 0.11 updaten
bekomme diese Fehlermeldung
`$ iobroker upgrade sun2000@0.11.0 --debug

Update sun2000 from @0.10.0 to @0.11.0

Installing iobroker.sun2000@0.11.0... (System call)

npm error code ENOTEMPTY

npm error syscall renamenpm error path /opt/iobroker/node_modules/modbus-serialnpm error dest /opt/iobroker/node_modules/.modbus-serial-xNQvrYX3npm error errno -39npm error ENOTEMPTY: directory not empty, rename '/opt/iobroker/node_modules/modbus-serial' -> '/opt/iobroker/node_modules/.modbus-serial-xNQvrYX3'

npm error A complete log of this run can be found in: /opt/iobroker/.npm/_logs/2024-07-08T05_41_35_268Z-debug-0.log

Could not handle ENOTEMPTY, because no deletable files were foundnpm error code ENOTEMPTYnpm error syscall renamenpm error path /opt/iobroker/node_modules/modbus-serialnpm error dest /opt/iobroker/node_modules/.modbus-serial-xNQvrYX3npm error errno -39npm error ENOTEMPTY: directory not empty, rename '/opt/iobroker/node_modules/modbus-serial' -> '/opt/iobroker/node_modules/.modbus-serial-xNQvrYX3'npm error A complete log of this run can be found in: /opt/iobroker/.npm/_logs/2024-07-08T05_41_35_268Z-debug-0.loghost.der-ripper Cannot install iobroker.sun2000@0.11.0: 217

ERROR: Process exited with code 25`

`root@der-ripper:/opt/iobroker/.npm/_logs# tail 2024-07-08T05_41_35_268Z-debug-0.log
34 error ENOTEMPTY: directory not empty, rename '/opt/iobroker/node_modules/modbus-serial' -> '/opt/iobroker/node_modules/.modbus-serial-xNQvrYX3'
35 silly unfinished npm timer reify 1720417295700
36 silly unfinished npm timer reify:retireShallow 1720417301491
37 verbose cwd /opt/iobroker
38 verbose os Linux 6.1.79-Unraid
39 verbose node v18.20.3
40 verbose npm  v10.8.1
41 verbose exit -39
42 verbose code -39
43 error A complete log of this run can be found in: /opt/iobroker/.npm/_logs/2024-07-08T05_41_35_268Z-debug-0.log
root@der-ripper:/opt/iobroker/.npm/_logs# `

liegt es an mir?
würd mich über ne kurze info sehr freuen

P.S.: die restlichen (damalichen) updates sind immer ohne probleme durchgerutscht.

### Version of nodejs

v18.20.3

### Version of ioBroker js-controller

v8.6.0

### Version of Adapter

0.10

### Repository

ioBroker stable

### Checklist of files to include below

- [ ] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

bolliy, 2024-07-08:
scheint ein Rechteproblem zu sein.
ich würde folgendes probieren:

iob stop             ## für iob fix muss iobroker gestoppt sein
iob fix
iob upgrade sun2000 
iob start

Stephan

luckyheiko, 2024-07-08:
für mich leider nicht möglich
[image]

bolliy, 2024-07-08:
ok, dein ioBroker läuft im Docker image! 
Ich habe gerade in meine Installation debian 11.9 (Raspberry Pi 4) auf 0.10.0 downgradet und danach auf 0.11.0 upgradet - ohne Rechteprobleme oder Fehlermeldung.
Im Docker image läuft ioBroker tatsächlich etwas anders. Da bin ich aber kein Experte und leider raus.
Da ich sonst von keinen Installationsproblemen gehört habe, würde ich auf das Docker Image tippen.

npm will modbus-serial updaten und das gelingt nicht, weil er die Rechte nicht hat.
Überprüfe mal die Rechte node_module Verzeichnis. Als Besitzer/Gruppe müssen iobroker iobroker sein. 

Stephan

bolliy, 2024-07-08:
Zu deiner Frage: npm will das installationsverzeichnis von dem paket modbus_serial umbennen. Danach müsste das Verzeichnis modbus_serial nicht mehr vorhanden sein. Das scheint nicht zu funktionieren.
Fehlermöglichkeiten:
* Rechteproblem
* Filesystem beschädigt
* eine andere Instanz (bash usw...) greift auf das Verzeichnis zu.

Dieser Vorgang wird von npm indirekt durch die eingestellten Abhängigkeiten des Adapter ausgelöst - aber nicht selbst vom Adapter eingeleitet oder durchgeführt.

Stephan

bolliy, 2024-07-11:
Konntest du das Upgrade-Problem lösen? 
Ich würde sonst gerne dieses issue schließen.

Stephan

luckyheiko, 2024-07-11:
lösen konnte ich das problem noch nicht.
liegt wohl am docker. (hoffe ich mal) und hier bin ich auf hilfe angewiesen.

von daher kannst erst mal schließen
https://forums.unraid.net/topic/101016-support-buanet-iobroker/page/8/#comment-1440029

bolliy, 2024-07-12:
versuch mal:

pkill -u iobroker
iobroker fix
-- Docker neustart --
iob upgrade sun2000

bolliy, 2024-07-14:
Ich habe gerade in einenm neuen docker image https://hub.docker.com/r/buanet/iobroker den Adapter sun2000 ohne Probleme installieren können.

luckyheiko, 2024-07-15:
bekomm es leider nicht hin
[image]

luckyheiko, 2024-07-15:
wenn ich aber auch eine 'neue' instanz hinzufüge, klappt das auch O.o
[image]

bolliy, 2024-07-15:
ok, konnte wohl leider nicht geheilt werden. Sonst musst du halt ein neues Docker image verwenden. 
Das funktioniert zumindestens.

LG Stephan