bolliy/ioBroker.sun2000 #116: Battery SOC Werte Atualisierung nach Sonneuntergang

Obsthaendler, 2024-10-01:


bolliy, 2024-10-01:
@Obsthaendler 
1. Welche sun2000 Adapter Version verwendest du?
2. Seit wann tritt der Fehler auf? 
3. Welcher deviceStatus haben die Inverter, sofern der Fehler auftritt?
4. Welcher RunningsStatus hat der Speicher, sofern der Fehler auftritt?
5. Sind alle Devices auf dem aktuellen Firmwarestand?

Obsthaendler, 2024-10-02:
Hi Stephan,

danke für deine Frage, dann versuche ich diese mal zu beantworte:

1. Welche sun2000 Adapter Version verwendest du?
12.1

2. Seit wann tritt der Fehler auf?
Hab erst seit letzter Woche den Speicher, daher seit dem, vorher waren keinerlei Fehlermeldungen vorhanden. Standby von Inverter 1 war immer ok. Inverter 0 lief auch durch.

3. Welcher deviceStatus haben die Inverter, sofern der Fehler auftritt?
Inverter 0: On-Grid, 
Inverter 1: Shutdown:fault

4. Welcher RunningStatus hat der Speicher, sofern der Fehler auftritt?
RUNNING

5. Sind alle Devices auf dem aktuellen Firmwarestand?
Ja, sind es. Dongle, Inverter, Batterie, MBUS

6. Ein Log-Auszug im Code-Block "<>". Hier ist besonders "The Inverter with modbus ID 2 switches to" von Interesse.
Wo finde ich das sun2000 log außerhalb von iobroker?

7. modelID der Inverter?
0: 10-KTL-M1
1: 15-KTL-M0

VG
Steffen

bolliy, 2024-10-02:
Hi Steffen,

dein Inverter 1 ist Augrund eines Fehlers oder beim Runterfahren in den Fehlerzustand gelaufen. Schau doch bitte mal nach den Alarm 1, Alarm 2 und Alarm 3 und vergleich das mit der Tabelle https://www.debacher.de/wiki/Sun2000_Modbus_Register.
GGf. ist beim Anschließen des Speichers etwas falsch gelaufen. Das würde auch die Fehlermeldung "Modbus exception 4: Slave device failure" im Log erklären.
LG Stephan

Obsthaendler, 2024-10-03:
Ich habe leider keine Lösung, ob die Verkabelung zwischen den Invertern und Speicher richtig sind, will ich nicht ausschließen, aber hab da keine Erfahrung. Ich habe heute die Inverter nochmals neu eingebunden und drauf geachtet ob die Kaskade (master/slave) stimmt. Im PV Forum liest man teils das auch der Slave nicht in Standby gehen darf. Ich bin gespannt, ob meine Neukonfiguration Änderungen heute Nacht bringt.

Alarm States reiche ich nach, hab nun erstmal Protokollierung eingeschaltet.

Rund um EVCC gibt es aber wohl derzeit auch modbus Batterie abfrage Probleme bei Huawei. Außerdem habe ich ja den neuen Huawei Speicher mit 7kwh. Vielleicht ist da was anderes.

bolliy, 2024-10-03:
Hallo Steffen,

vielen Dank für deine Spende. Ich habe mich sehr gefreut.

Zu deinem Speicherabfrageproblem:
Der 2te WR dürfte eigentliche erst einen shutdown machen, sofern der angeschlossene Speicher auf MinSOC entladen ist. Sonst kann er die Lasten im Haus nicht mit Wechselstrom bedienen.
Du kannst das "Einschlafen" deines Wechselrichter durch einfaches Setzen des Wertes verhindern.
`sun2000.0.inverter.1.control.battery.chargeFromGridFunction = true`

Die Alam States findest du einfach in den Object states:
```
sun2000.0.inverter.1.alarm1
sun2000.0.inverter.1.alarm2
sun2000.0.inverter.1.alarm3
```
LG Stephan

Obsthaendler, 2024-10-03:
Die Werte innerhalb von x.control sind alle NULL, im Adapter habe ich aber Battery Control aktiviert. Wieso?
Vielleicht stimmt ja doch was nicht mit der Kommunikation zwischen WR und Speicher.

Obsthaendler, 2024-10-04:
Beim setzen von sun2000.0.inverter.1.control.battery.chargeFromGridFunction = true erscheint folgendene Meldung

```
sun2000.0 | 2024-10-04 08:56:54.208 | info | Control: Event is discarded because it could not be processed. inverter.1.control.battery.chargeFromGridFunction
-- | -- | -- | --
sun2000.0 | 2024-10-04 08:56:54.207 | warn | Error while writing to 192.168.2.16 [Reg: 47087, Len: 1, modbusID: 2] with: Modbus exception 2: Illegal data address (register not supported by device)
sun2000.0 | 2024-10-04 08:56:31.619 | warn | Error while writing to 192.168.2.16 [Reg: 47087, Len: 1, modbusID: 2] with: Modbus exception 2: Illegal data address (register not supported by device)
sun2000.0 | 2024-10-04 08:56:27.812 | info | Control: Event - state: battery.chargeFromGridFunction changed: true ack: false
```

Die Alarmzustände ändern sich garnicht.

Obsthaendler, 2024-10-05:
Es liegt scheinbar nur daran das Inverter.1 offline ist. Hat er den Status Standby ist es ok, geht er komplett offline dann kommen die Fehler. 

Gestern Abend habe ich in der Einstellung dann nur Modbus ID 1  (inverter.0 = Master) abgefragt, dann passten alle Werte der Battery. Meine Frage daher generell, wieso gibt es bei Inverter 1 überhaupt Battery States? Das macht in der Regel doch alles der Master der immer online ist. 
 
Alle anderen Inverter übermitteln doch nur ihre Power. Ohne aktiven Speicher war vorher übrigens alles ok mit der Standby/Offline Erkennung. Dazu hatte ich ganz am Anfang mal mit dir gesprochen.  

VG
Steffen

bolliy, 2024-10-05:
Hallo Steffen,
nun bin ich etwas irritiert. Ich hatte es so verstanden, dass dein Speicher am Slave und nicht am Master angeschlossen ist.
Ist dein Speicher am Master oder Slave angeschlossen?

Die Daten werden immer über Master (Sdongle) abgefragt, das ist richtig. Sofern es die Registerdaten vom Slave betreffen werden diese Anfragen über RS485 Kabel vom Master an den Slave weitergeleitet.
Falls am Slave kein Speicher angeschlossen ist, sollte dies vom Adapter erkannt werden und keine Speicherabfragen erfolgen. Dazu wird beim Starten des Adapters die SN der Battery Units als Indikator ausgelesen: https://github.com/bolliy/ioBroker.sun2000/blob/bec37dddbcf7541b8a6a94586ad379b53086a240/lib/drivers/driver_inverter.js#L233

LG Stephan

bolliy, 2024-10-05:
Ich habe nochmal unsere Unterhaltung durchgelesen und denke ich bin von einer falschen Konfiguration ausgegangen.

Sofern meine Annahme nun richtig ist, dann wird dein **Speicher am Master** angeschlossen sein - was ja auch naheliegend ist.

Dann sind es wohl 2 unterschiedliche Probleme, die ggf. in keinem kausalen Zusammenhang stehen:
1. die SOC Werte des Speichers am Master werden nicht aktualisiert
2. der Slave geht in "shutdown: fault"

Zu 1: kann ich nicht nachvollziehern, da der Master "On-Grid" ist! Im Zustand "On-Grid" werden immer alle Register gelesen. Das müsste noch noch genauer untersucht werden.

Zu 2: den Zustand "shutdown: fault" habe ich tatsächlich in meiner Programmierung nicht berücksichtigt. Ich denke aber, dass ich in diesem Zustand keine Anfragen an den Slave stellen darf. Das werde ich mal umsetzen ... und dir eine Entwicklker-Version bauen...

LG Stephan

Obsthaendler, 2024-10-05:
Test Version ist installiert. Ich gebe dir eine ausführliche Rückmeldung morgen dann. Bin grad unterwegs. 
Aber deine Annahmen sind alle korrekt. 
Vielleicht sollte man eine Issue Vorlage erstellen die Leute nach ihrer Config fragen sodass man die Ausgangslage kennt.

Obsthaendler, 2024-10-07:
Nach nun mehr als 24h kann ich sagen, dass die Testversion gut läuft. Nachts (zwischen Sonnenuntergang und Aufgang) habe ich jedoch im Log alle 10 Sekunden Conntected Modbus ... Open Connection. Keine Ahnung wieso, allerdings werden alle Werte richtig verarbeitet. Der Battery SOC wird exakt ausgelesen, auch wenn der Speicher auf 5% ist, welches das Minimum ist kommen keine Fehler hinzu.

Was aber irritierend ist, wieso in der Objektestruktur weiterhin beim Slave Battery Objekte vorhanden sind. Habe diese gelöscht und den Adapter neu gestartet. Sie werden weiterhin neu angelegt. Wie du geschrieben hast, ist der Speicher am Master, genau wie der Dongle.  Was in der Regel die Standard Vorgabe von Huawei ist. 

Ich habe nun beim Device.Status von Inverter.1 (Slave) kein "Shutdown: Fault" mehr, aber "Standby: detecting" bis "Standby: initializing" obwohl FusionSolar sagt, er seie Offline, der letzte Status ist dann "Standby: initializing". Wie fragst du den Offline Stauts ab?
Ich denke wenn die Abfrage zur Erkennung sauber funktioniert, kann man viele Probleme beheben. Hierzu bräuchte man mal andere Meinungen mit ähnlichen Aufbau.

Danke für die schnelle Anpassung und Hilfe.

VG
Steffen

bolliy, 2024-10-07:
Hallo Steffen,
ich denke die Wiederherstellung der Verbindung (Conntected Modbus …) wird vom Adapter durchgeführt, da es zu Verbindungsfehlern bei der Statusabfrage des Slave WRs kommt. 
Der Status wird über die Registeradresse 32089 abgefragt. Die unterschiedlichen Status sind für den „normalen“ User eher verwirrend. Deshalb wird der WR als „Offline“ deklariert.

Ich werde versuchen den Battery Path nicht aufzubauen, sofern kein Speicher vorhanden ist. 

Sofern wieder etwas zu testen gibt, werde ich mich hier melden....

Stephan

bolliy, 2024-10-08:
Hi Steffen,
ich habe nun versucht die Meldungen "Conntected Modbus ..." zu umgehen, da ein Neufaufbau der Verbindung bei modbusCode=4 eigentlich nicht notwendig sein sollte. Der Path "battery" sollte nun beim Slave WR nicht mehr aufgebaut werden. Bitte den Pfad vorher löschen.
Installation wie oben beschrieben.
Kannst ja mal berichten. 

PS: In der Entwicklung ist auch die Integration der Huawei **EMMA** und diese befindet sich nun unter dem Reiter "Integration". Der SmartLogger ist dann unter diesem Reiter verschoben worden.
  
 LG Stephan

Obsthaendler, 2024-10-10:
Adapter wurde aktualisiert. Battery erscheint beim Slave nicht mehr. Abwarten was nach Sonnenuntergang passiert. Danke

VG
Steffen

Obsthaendler, 2024-10-11:
Es erscheinen keine derartigen Warnungen mehr im Log, daher schließe ich hiermit mein Probkem. Ich denke du wirst es bei passender Gelegenheit in eine neue Version mit einbauen. Danke.

VG
Steffen

closed 2024-10-11 by Obsthaendler