bolliy/ioBroker.sun2000 #119: Werte über Modbusproxy zurück schreiben lassen

Obsthaendler, 2024-10-14:
Hallo,
eine nützliche Funktion wäre es, wenn über den Modbusproxy auch Werte geschrieben werden könnten. Z.B. bei Nutzung von evcc  wird eine aktive Batteriesteuerung genutzt, diese verhindert das entladen der Batterie. (https://docs.evcc.io/docs/features/battery#aktive-batteriesteuerung)

In meinem Fall wird hierbei das Register 47100 genutzt ([Energy storage] Forcible charge / discharge)

Es müsste daher die Anfrage im Modbusproxy zu aller erst registriert werden, ob der Wert ankommt und dann weiterverarbeitet wird.

VG
Steffen

bolliy, 2024-10-16:
@Obsthaendler Hallo Steffen!
Zuerst muss ich einen Async-Lock für das Serialisieren der Datenanfrage an den WR programmieren. Ansonsten kann ich keine synchronen Responses auf die Schreibvorgänge geben. Das ist ein bisschen kniffelig, da die Huawei modbus Schnittstelle des SDongles leider keine gleichzeitigen Anfragen verträgt. 

LG Stephan

bolliy, 2024-10-20:
Steffen, bitte die v0.14.0 installieren. Sofern die Umsetzung ok ist, bitte das issue schleißen.

Stephan

Obsthaendler, 2024-10-22:
Bin mir nicht sicher... aber folgendes Log habe ich erhalten, nachdem über den Modbusproxy via evcc hätte was geschrieben werden müssen:

Um 08:19 bzw. paar Sekunden später wurde ein Befehl (Batterie sperren) empfangen, hier gab es dann ein Reconnect und um 8:21 als ich wieder zurück auf "normal" bin, wieder reconnect. 

`
sun2000.0 | 2024-10-22 08:22:45.113 | info | Modbus tcp server: {"stat":{"#WaitForConnected-address_32064-value_2-unidId_1":5,"#WaitForConnected-address_32064-value_2-unidId_2":5,"#WaitForConnected-address_37001-value_2-unidId_1":4,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":2933,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":2933,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":2933,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":2933,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":2933,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":2933,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":2933,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":2933,"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":2932,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":2932,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":2932,"#getMultipleHoldingRegisters-address_32106-value_2-unidId_2":32,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":2900}}
-- | -- | -- | --
sun2000.0 | 2024-10-22 08:22:17.043 | info | Connected Modbus TCP to 192.168.2.16:502
sun2000.0 | 2024-10-22 08:22:12.039 | info | Open Connection...
sun2000.0 | 2024-10-22 08:22:12.037 | warn | Error while reading from 192.168.2.16 [Reg: 37113, Len: 2, modbusID: 1] with: Timed out
sun2000.0 | 2024-10-22 08:21:45.114 | info | Modbus tcp server: {"stat":{"#WaitForConnected-address_32064-value_2-unidId_1":5,"#WaitForConnected-address_32064-value_2-unidId_2":5,"#WaitForConnected-address_37001-value_2-unidId_1":4,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":2931,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":2931,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":2931,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":2931,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":2931,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":2931,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":2931,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":2931,"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":2930,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":2930,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":2930,"#getMultipleHoldingRegisters-address_32106-value_2-unidId_2":32,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":2898}}
sun2000.0 | 2024-10-22 08:21:45.113 | warn | Not all data can be read! Please inspect the sun2000 log.
sun2000.0 | 2024-10-22 08:21:12.376 | info | Connected Modbus TCP to 192.168.2.16:502
sun2000.0 | 2024-10-22 08:21:07.371 | warn | Modbus-proxy: [object Object]
sun2000.0 | 2024-10-22 08:21:07.370 | info | Open Connection...
sun2000.0 | 2024-10-22 08:20:45.113 | info | Modbus tcp server: {"stat":{"#WaitForConnected-address_32064-value_2-unidId_1":5,"#WaitForConnected-address_32064-value_2-unidId_2":5,"#WaitForConnected-address_37001-value_2-unidId_1":4,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":2927,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":2927,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":2927,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":2927,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":2927,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":2926,"#getMultipleHoldingRegisters-address_32106-value_2-unidId_2":32,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":2894}}
sun2000.0 | 2024-10-22 08:20:45.112 | warn | Not all data can be read! Please inspect the sun2000 log.
sun2000.0 | 2024-10-22 08:20:33.751 | info | Connected Modbus TCP to 192.168.2.16:502
sun2000.0 | 2024-10-22 08:20:28.744 | warn | Error while reading from 192.168.2.16 [Reg: 32080, Len: 2, modbusID: 1] with: Timed out
sun2000.0 | 2024-10-22 08:20:28.744 | info | Open Connection...
sun2000.0 | 2024-10-22 08:20:14.291 | info | Connected Modbus TCP to 192.168.2.16:502
sun2000.0 | 2024-10-22 08:20:09.287 | warn | Modbus-proxy: [object Object]
sun2000.0 | 2024-10-22 08:20:09.286 | info | Open Connection...
sun2000.0 | 2024-10-22 08:19:59.284 | info | Connected Modbus TCP to 192.168.2.16:502
sun2000.0 | 2024-10-22 08:19:54.280 | warn | Error while reading from 192.168.2.16 [Reg: 32064, Len: 2, modbusID: 1] with: Timed out
sun2000.0 | 2024-10-22 08:19:54.280 | info | Open Connection...
sun2000.0 | 2024-10-22 08:19:45.111 | info | Modbus tcp server: {"stat":{"#WaitForConnected-address_32064-value_2-unidId_1":5,"#WaitForConnected-address_32064-value_2-unidId_2":5,"#WaitForConnected-address_37001-value_2-unidId_1":4,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":2926,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":2926,"getMultipl [... cut ...] ddress_37109-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":2926,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":2925,"#getMultipleHoldingRegisters-address_32106-value_2-unidId_2":32,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":2893}}
`

bolliy, 2024-10-22:
Da scheint noch nicht zu funktionieren. Werde das Logging nochmal verbessern, damit ich sehen kann was dort geschrieben wird.
Melde mich ...
LG Stephan

bolliy, 2024-10-22:
bitte wieder über github installieren. Version erhöht sich nicht.
Ggf. muss der Adapter neu gestarted werden. 
Jetzt wartet der proxy bis der Adapter connected ist mit dem Schreiben von Werten an den WR und es gibt aussagekrätigere Fehlermeldungen.
Stephan

Obsthaendler, 2024-10-22:
Habe ich gemacht. Jetzt hatte es auch irgendwie funktioniert. Vielleicht ist es weiterhin ein persönliches Problem meiner Konfiguration mit dem 2ten WR.

10:12 Batteriesperre
10:17 Aufheben

sun2000.0 | 2024-10-22 10:16:51.087 | info | Modbus tcp server: {"stat":{"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":13,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":13,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":13,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":13,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":13,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":13,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":13,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":13,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":13,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":13,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":13,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":13,"setRegister-address_47100-value_1-unidId_1":19,"setRegister-address_47246-value_1-unidId_1":19,"setRegister-address_47083-value_1-unidId_1":19,"setRegisterArray-address_47249-value_2-unidId_1":19}}
-- | -- | -- | --
sun2000.0 | 2024-10-22 10:15:51.086 | info | Modbus tcp server: {"stat":{"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":11,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":11,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":11,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":11,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":10,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":10,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":10,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":10,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":10,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":10,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":10,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":10,"setRegister-address_47100-value_1-unidId_1":15,"setRegister-address_47246-value_1-unidId_1":15,"setRegister-address_47083-value_1-unidId_1":15,"setRegisterArray-address_47249-value_2-unidId_1":14}}
sun2000.0 | 2024-10-22 10:14:51.085 | info | Modbus tcp server: {"stat":{"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":9,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":9,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":9,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":9,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":9,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":9,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":9,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":9,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":9,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":9,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":9,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":9,"setRegister-address_47100-value_1-unidId_1":11,"setRegister-address_47246-value_1-unidId_1":11,"setRegister-address_47083-value_1-unidId_1":11,"setRegisterArray-address_47249-value_2-unidId_1":11}}
sun2000.0 | 2024-10-22 10:13:51.086 | info | Modbus tcp server: {"stat":{"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":7,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":7,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":7,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":7,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":7,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":7,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":7,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":7,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":7,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":7,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":7,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":7,"setRegister-address_47100-value_1-unidId_1":7,"setRegister-address_47246-value_1-unidId_1":7,"setRegister-address_47083-value_1-unidId_1":7,"setRegisterArray-address_47249-value_2-unidId_1":7}}
sun2000.0 | 2024-10-22 10:12:51.085 | info | Modbus tcp server: {"stat":{"getMultipleHoldingRegisters-address_32064-value_2-unidId_1":5,"getMultipleHoldingRegisters-address_32106-value_2-unidId_1":5,"getMultipleHoldingRegisters-address_32064-value_2-unidId_2":5,"getMultipleHoldingRegisters-address_32106-value_2-unidId_2":5,"getMultipleHoldingRegisters-address_37001-value_2-unidId_1":5,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":5,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":5,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":5,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":5,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":5,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":5,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":5,"setRegister-address_47100-value_1-unidId_1":3,"setRegister-address_47246-value_1-unidId_1":3,"setRegister-address_47083-value_1-unidId_1":3,"setRegisterArray-address_47249-value_2-unidId_1":3}}
[... cut ...] ltipleHoldingRegisters-address_37001-value_2-unidId_1":3,"#getMultipleHoldingRegisters-address_37068-value_2-unidId_1":3,"getMultipleHoldingRegisters-address_37004-value_1-unidId_1":3,"getMultipleHoldingRegisters-address_37113-value_2-unidId_1":3,"getMultipleHoldingRegisters-address_37107-value_2-unidId_1":3,"getMultipleHoldingRegisters-address_37109-value_2-unidId_1":3,"getMultipleHoldingRegisters-address_37111-value_2-unidId_1":3,"getMultipleHoldingRegisters-address_37121-value_2-unidId_1":3}}

bolliy, 2024-10-22:
Steffen bitte wieder über github installieren. Version erhöht sich nicht.
Ggf. muss der Adapter neu gestarted werden.

Nun wird beim Schreiben über den modbus-proxy ebenfalls in den read-cache geschrieben. Mir ist aufgefallen, das evcc sehr viele gleiche Schreibvorgänge macht. Ich denke, dass evcc nach dem Schreiben gleich nachschaut ob der Wert angekommen ist. 
Jetz wird der Lesevorgang gleich aus dem Cache mit dem "richtigen" Wert geantwortet. Das sollte das Phänomen etwas "entprellen"...
LG Stephan

bolliy, 2024-10-24:
Hi Steffen, funktiniert jetzt alles?
Ich würde sonst gerne eine neu Version publishen. 

LG Stephan

closed 2024-10-24 by Obsthaendler