bolliy/ioBroker.sun2000 #120: [Bug]: Leistungswerte einzelner Batteriepacks fehlen in den Datenpunkten

severendis, 2024-10-15:
### The problem

Leistungswerte einzelner Batteriepacks fehlen in den Datenpunkten

In der Konstellation mit mehreren Wechselrichtern an denen jeweils mehrere LUNA Batteriespeicher angeschlossen sind, fehlen die einzelnen Leistungen für Ladung und Entladung je LUNA (also je Leistungsmodul).
Im Anhang ein Bild in dem die Daten fehlen. Es gibt je UNIT z.B. die eigene Temperatur, aber nicht die Lade-/und Entladeleistung oder den Strom. Diese gibt es nur kumuliert je Wechselrichter.
Es geht zum Beispiel über die Stromangabe je Leistungsmodul:

Die ersten beiden Leistungsmodule die an Wechselrichter 1 hängen:
Unit ID 1 / 37001
Unit ID 1 / 37743

Die nächsten beiden Leistungsmodule, die an Wechselrichter 2 hängen:
Unit ID 2 / 37001
Unit ID 2 / 37743

Ich habe noch nicht geschaut, ob es die Leistung in Watt auch gibt.
Wäre schön, die Werte je Leistungsmodul noch im Adapter zu bekommen.

### Version of nodejs

20.18.0

### Version of ioBroker js-controller

6.0.11

### Version of Adapter

0.11.0

### Repository

ioBroker stable

### Operating system running ioBroker

Linux

### Checklist of files to include below

- [ ] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

[image]

bolliy, 2024-10-16:
@severendis bitte 0.13.0 (beta) [installieren](https://github.com/bolliy/ioBroker.sun2000/wiki/Adapter-installieren). 
Dort wird für jedes Battery Pack die Lade/Entladeleistung angegeben. Unter dem Reiter "Further Register" die Battery Packs anhaken.

Stephan

bolliy, 2024-10-18:
@severendis falls dein Anliegen erledigt ist, bitte das issue schließen.
Stephan

bolliy, 2024-10-20:
Since the author of this request does not respond, I consider this to be resolved!

closed 2024-10-20 by bolliy

severendis, 2024-10-21:
Danke für das Update. Ich habe gerade die 0.14.0 installiert.
Ich sehe jetzt die Werte der einzelnen Batteriepacks.
Es könnte jetzt auch in der Ebene darüber den Wert je BCU geben. Also in der Ebene in der auch der SOC der kompletten LUNA steht.
Aber sonst sehr gut gelöst. 
Vielen Dank.