bolliy/ioBroker.sun2000 #161: [Feature Request]: integration eines zweiten SmartLoggers

PrinzEisenherz200976, 2025-03-12:
### What would you like to be added?

Neben einer bestehenden Anlage (SUN2000 10ktl-m1, LUNA2000 10kW, DTSU666-H 3-Phasig) noch ein Balkonkraftwerk mit Hoyemiles-Wechselrichter am laufen. Damit ich die Daten des Hoymiles auch auf FusionSolar sehe habe ich zusätzlich einen DDSU666-H (1-Phasig) installiert. Dieser läuft auf ModBus-ID 19.
Der DTSU666 läuft auf ModBus-ID 11.
Jetzt würde ich gerne mit diesem Adapter auch auch die die Daten des DDSU666 auslesen.
Besteht die Möglichkeit eine Abfrage für einen zweiten SmartLogger zu integrieren, wo ich die Modbus-ID manuel einstellbar ist?
Mein System ist nach folgendem Schema konfiguriert:
[Vernetzung mit Wechselrichtern anderer Hersteller](https://support.huawei.com/enterprise/de/doc/EDOC1100173562/c79b0469)

### Why is this needed?

Das Auslesen des DTSU666 klappt ja mit dem Adapter wunderbar. Jetzt wäre es schön wenn zusätzlich noch ein zweiter SmartLogger ausgelesen werden könnte!

Gruß
Johnny

bolliy, 2025-03-14:
@PrinzEisenherz200976 ich habe mal im Programmcode von edyso geschaut. Der Smart Meter wird bei ihm per serial modbus und nicht per tcp abgefragt. Siehe hier https://github.com/edyso/ioBroker.sun2000/blob/main/lib/drivers/driver_rtumeter.js

PrinzEisenherz200976, 2025-03-16:
Hi Bolliy, sorry für die späte Antwort!
Ich habe mich da falsch ausgedrückt. Ich meinte natürlich nicht den SmartLogger sondern die SmartPowerMeter.
Wie fragt denn Dein Adapter folgende Werte ab?

[image]

Hier sind die Werte ohne den Einfluss des DDSU666 vom Balkonkraftwerk.

Und hier sind die Werte vom DTSU und DDSU zusammengefasst. Also was von der 10kW Anlage und vom Balkokraftwerk kommt. 

[image]

Der SDongle ermittelt ja irgendwie die Daten. Bezieht er die Daten vom SmartPowerMeter? Holt er die sich von der vordefinierten ModBus-ID? 

Gruß
Johnny

bolliy, 2025-03-16:
was unter sDongle steht wird aus dem sDongle mit der modbus id 100 gelesen. Hier die Definitions https://github.com/bolliy/ioBroker.sun2000/blob/main/docs/definitions.md

Das 2te Bild sind die Daten aus dem WR mit der angegbenen modbus id (meist 1).

Aus dem Vergleich bzw. Differenz der Daten sun2000.0.sdongle.totalInputPower und sun2000.0.collected.inputPower sollte sich doch die Leistung deines Balkonkraftwerks ergeben?

Stephan

bolliy, 2025-03-16:
Allerdings werden die Daten in einem unterscheidlichen Intevall abgefragt - was den Vergleich bei Wolken am Himmel etwas schwieriger macht: https://github.com/bolliy/ioBroker.sun2000/wiki/Verarbeitungsintervall-(interval)

PrinzEisenherz200976, 2025-03-16:
Ich denke, ich werde das ganze mit nem Script jetzt umsetzen. Wenn es nicht 100% genau ist, ist es auch nicht so schlimm. Ich hätte nur gehofft das ich mir ein zusätzliches Script sparen könnte wenn es der Adapter macht. Trotzdem vielen Dank das du es dir mal angeguckt hast. Der Request kann geschlossen werden.
Schönen Abend noch

closed 2025-03-17 by bolliy