bolliy/ioBroker.sun2000 #176: [Feature Request]: Registernummer 40126

schneewa, 2025-06-30:
### What would you like to be added?

Wäre es möglich die Registernummer "40126" - Ausgangsleistung des Wechselrichter (RW) in den Adapter auf zu nehmen.

besten Dank

### Why is this needed?

Dies ist wichtig um die Ausgangsleistung des Wechselrichters zu steuern

bolliy, 2025-06-30:
Ich habe die Register 40037 - 40198 (Power grid scheduling) bisher ausgelassen, da ich nicht genau weiß wie man die richtig einsetzt.
Vielleicht kannst du mich bitte etwas aufklären.

schneewa, 2025-07-01:
Hi, danke für die schnelle Rückmeldung

Man kann mit diesen Registern den Wechselrichter steuern 
z.B.: mit den 40126 kann man, wenn man nicht die gesamte verfügbare Leistung einspeisen möchte, die Leistung des Wechselrichters reduzieren – dies ist notwendig, wenn der Einspeisetarif negativ wird

ist für meine Anwendung besonders wichtig – ich denke auch, dass das andere benötigen

Mit der Bitte um Einbau dieser Power Grid Register in die Instanz

Lg und besten Dank

bolliy, 2025-07-01:
Würdest du dann im Gegenzug ein kleine Beschreibung schreiben, damit wir der Community aufzeigen können warum gerade dieses Register anstatt des Export Control für die Reduzierung der Einspeiseleistung wichtig ist? Ich möchte das gerne im Wiki hinterlegen.
Sonst steigt bald keiner mehr durch die vielen Einstellmöglichkeiten durch.

LG Stephan

schneewa, 2025-07-01:
nur das wir vom selben sprechen - du meinst die Reg. Nr. 47415  (Active power control mode) in der Kombination mit 47416 (Maximum Feed Grid Power (kW)

schneewa, 2025-07-01:
Habe ich auch selbst ausprobiert - der WR geht sofort auf 0kW oder 0% Einspeisung - wenn kein Smart Power Sensor vorhanden ist - der Wert vom Register 47416 wird ignoriert

schneewa, 2025-07-01:
Hi vielen Dank - Der Wert ist leider Schreibgeschütz ausgeführt

schneewa, 2025-07-01:
Funktioniert - vielen Dank

lg

bolliy, 2025-07-02:
ich habe gerade noch den prozentualen gid power auf github eingebaut.

Stephan

closed 2025-07-06 by schneewa