bolliy/ioBroker.sun2000 #190: grid Export und Import tauschen

bahnuhr, 2025-09-21:
### The problem

Hallo,
wie hier beschrieben:
https://forum.iobroker.net/topic/71768/test-adapter-sun2000-v0-1-x-huawei-wechselrichter/1036?_=1758432236926
stimmen die DP nicht.

Sie müssen genau umgedreht angelegt werden.
mfg
Dieter

### Version of Adapter

2.1.0

### Repository

ioBroker stable

### Checklist of files to include below

- [ ] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

bolliy, 2025-09-21:
Hi Dieter, die Emma hat anscheinend eine umgekehrte Vorzeichenlogik bei allen Meter-Leistungswerten.
Da ich leider keine Emma im Einsatz habe müsstest du bitte einige weitere Wert auf das Richtigkeit überprüfen:

Ist das Vorzeichen richtig?
sun2000.0.meter.activePowerL1 
sun2000.0.meter.activePowerL2 
sun2000.0.meter.activePowerL3 

und die beiden Werte müssten vertauscht sein?
sun2000.0.meter.positiveActiveEnergy 
sun2000.0.meter.reverseActiveEnergy 

Stephan

bahnuhr, 2025-09-22:
sun2000.0.meter.activePower ist aktuell negativ
aber
sun2000.0.meter.activePowerL1
sun2000.0.meter.activePowerL2
sun2000.0.meter.activePowerL3
sind alle positiv.

Und vertauschen musst du m.E.:
sun2000.0.collected.gridExportStart
sun2000.0.collected.gridExportToday
mit
sun2000.0.collected.gridImportStart
sun2000.0.collected.gridImportToday

Also nicht bei meter; sondern bei collected.

bahnuhr, 2025-09-24:
Habe über katze und git installiert.
Installiert wurde die 2.1.1.
Wo finde ich die 2.2.1 ?

bolliy, 2025-09-24:
Habe mich verschrieben. Die 2.1.1 ist richtig

bahnuhr, 2025-09-25:
so:
bei collected hattest du ja gridex und gridim vertauscht.
Diese Werte passen auch.
Bei meter passen die Werte aber noch nicht.

[image]

Wie du siehst muss die Bezeichnung bei meter genau anders rum sein.

bolliy, 2025-09-25:
Ich glaube die Behandlung der Meter-Daten erfolgt beim sDongle und der Emma anders. Beim sDongle haben die Leistungswerte einen positiven Wert, wenn Strom ins das Netz eingespeist werden. Bei Strombezug einen negative Leistungswerte. Bei der Emma vermute ich die umgekehrte Sichtweise. Deshalb wurde bereits bei der ersten Version des Adapters mit der Emma-Unterstützung das Vorzeichen von meter.activePower beanstandet. So wurde das Vorzeichen von activePower umgedreht und nun auch positiveActiveEnergy mit reverseActiveEnergy getauscht. 
Eigentlich müssen die Vorzeichen der Meter-Werte unverändert bleiben. Die nachgelagerte Berechnung der Collected-Werte müssen dann entsprechend der verbauten Hardware unterschiedlich implementiert werden.

Da ich keine Emma besitze, bitte ich dich die Werte in der FusionSolar App mit meinen zu vergleichen. Siehe obiger Hinweis. Ich habe tatsächlich mehr inst Netz eingespeist als bezogen.

[image]
[image]

Stephan

bahnuhr, 2025-09-28:
Konntest du dies schon einmal anschauen ?

bolliy, 2025-09-28:
Ich führe gerade eine neuen Datenpunkt **directionOfPosivitePowerFlow** ein, diese soll die Richtung des positiven Stromflusses festlegen. Ich bin also bei der Umsetzung....

bolliy, 2025-09-29:
"Dann müssten doch die DP untereinander passen." Das sehe ich auch so. 
Aber die Emma (built-in electric energy sensor)  liefert im Gegensatz zum SDongle eben leider nicht die vergleichbaren Daten!
So wurde bereits früh von einem User bemerkt, dass das Vorzeichen beim DP sun2000.0.meter.activePower nicht stimmen könne. So habe ich das Vorzeichen angepasst. 
Ich möchte aber die Daten in der Art und Weise darstellen, wie diese über modbus gelesen werden. So habe ich mich entschieden die Unterschiedlichkeit im DP directionOfPowerFlow aufzuzeigen.
Damit man meter.ActivePower nicht umrechnen (Änderung des Vorzeichens) muss, gibt es nun den neuen DP sun2000.0.collected.feed-inPower - also die Einspeiseleistung.

bahnuhr, 2025-09-30:
Hallo,
neu von git installiert.
Jetzt stimmt gar nichts mehr !
Offensichtlich hast du wieder collected.gridimport mit export getauscht.
Die Werte stimmen so nicht.

Meine Hinweise scheinen nicht gewollt zu sein.
Ich bin jetzt raus.

Werde mich mal umschauen wie ich zu sinnvollen Werten komme.

[image]

bolliy, 2025-10-02:
@bahnuhr ich glaube den Fehler gefunden zu haben. Kannst per github installieren. (Neustart nicht vergessen!)
Die gridExportToday und gridImportToday werden sich erst um Mittenacht korregieren. Danach sollten die Werte aber stimmen.

Stephan

bahnuhr, 2025-10-03:
Nein, der Fehler ist schlimmer geworden.
Wie du siehst:

[image]

negativer Wert bei gridImportToday kann nicht sein.

bolliy, 2025-10-03:
@bahnuhr vielen Dank für deine Beobachtungen.
Jetzt sehe ich einen Übertragungsfehler in der Emma-Umsetzung. Den angefragten Cache-Wert gibt es nicht. Deshalb ist der Today-Wert negativ. 
Jetzt bitte nochmal per github installieren....

bolliy, 2025-10-04:
Ich habe nochmal alle genannte Punkte eingebaut. Würdest du nochmal final die github Version testen? 
Sofern du nichts mehr findest, würde ich eine neue npm Version bereit stellen.
Stephan

bahnuhr, 2025-10-04:
Na klar, ich schau es mir an.
Mal schauen, ob ich noch was finde.

bahnuhr, 2025-10-04:
[image]

Ist dies jetzt doppelt ?
Nochmal:
Ist das nicht verwirrend.
Warum greifst du die Einstellung nicht von der Instanz ab. Denn dort steht doch drin ob es Emma oder dongle ist.

bahnuhr, 2025-10-04:
für was ist:
collected.surplus ?

Da ändert sich nix.

bolliy, 2025-10-04:
den DP directionOfPowerFlow löschen. Wird nicht mehr benötigt. Das kein Release für die Änderungen gab, werden diese nicht auto. gelöscht. Ja könnte man aus der Instanz abgreifen. Die Variante finde ich aber besser!

bolliy, 2025-10-04:
https://github.com/bolliy/ioBroker.sun2000/wiki/%C3%9Cberschuss-(surplus)

bahnuhr, 2025-10-04:
surplus
ok, muss ich mir mal anschauen.
So auf die Schnelle hab ichs nicht kapiert.

bahnuhr, 2025-10-04:
sieht in VIS dann so aus:

[image]

bahnuhr, 2025-10-04:
Habe mal schnell ein kleines Script geschrieben:
`on({id: 'sun2000.0.collected.inputPower', change: 'any'}, function(obj) {
    // 'obj' enthält Informationen über die Änderung, wie die neue ID und der Wert (obj.state.val)
    log('Der Datenpunkt ' + obj.id + ' hat sich geändert. Neuer Wert: ' + obj.state.val);

    log ("0= " + getState("sun2000.0.inverter.0.inputPower").val + ", 1= " + getState("sun2000.0.inverter.1.inputPower").val + ", 2= " + getState("sun2000.0.inverter.2.inputPower").val);
    log ("gesamt = " + (getState("sun2000.0.inverter.0.inputPower").val + getState("sun2000.0.inverter.1.inputPower").val + getState("sun2000.0.inverter.2.inputPower").val));

    var houseconsum1 = getState("sun2000.0.inverter.0.activePower").val + getState("sun2000.0.inverter.1.activePower").val
        + getState("sun2000.0.inverter.2.activePower").val - getState("sun2000.0.meter.derived.feed-inPower").val;
    log ("hc1= " + houseconsum1);

    var houseconsum2 = getState("sun2000.0.collected.activePower").val - getState("sun2000.0.meter.derived.feed-inPower").val;
    log ("hc2= " + houseconsum2);

});`

Und raus kommt z.B.:

[image]

Fragen:
müsste nicht hc1 und hc2 das gleiche sein.

Und dies ist das, was ich meine.
Es passt halt mathematisch nicht zusammen.
Wenn ich nun noch Netz dazu darstelle, dass passt dies mathematisch nicht.

bolliy, 2025-10-04:
1. Die Daten haben leider nicht den gleichen Zeitstempel. Der modbus Implementierung von Huawei ist recht langsam.
2. Du liest getState("sun2000.0.meter.derived.feed-inPower").val mehrmals, was zu unterschiedlichen Ergebnissen führen kann. Besser eine Variable verwenden.  

Um dem Problem auf die Speur zu kommen würdeich erstmal die activePower der Inverter mit collected vergleichen. Dort wird es sicherlich eine Diskrepanz ergeben.

bahnuhr, 2025-10-04:
> 1. Daten haben leider nicht den gleichen Zeitstempel

genau, und deshalb ist das für mich nix.
Ich werde (wie bisher) das manuell per Script berechnen, denn
Produktion +- Batterie +- Netz muss mathematisch Haus-Verbrauch ergeben.

Und wenn die Werte halt unterschiedlich rein fließen, dann passt es halt mathematisch nicht. Und dies kommt halt bei VIS nicht gut an.